kercre123/wire-pod #2: sudo ./setup.sh not generating any files

kakashi2018, 2022-08-10:
getting this error at the end

 Created source.sh file!

./setup.sh: line 455: cd: certs: No such file or directory
Creating server_config.json for robot
cat: address: No such file or directory
Created!

I'm trying this on raspberry pi3b and raspberry pi os 64 bit.

kercre123, 2022-08-10:
IIRC those errors are normal for now if you are using the escapepod.local option, since it uses pre-existing certs and an already-known address, and it hasn't been accounted for in setup.sh yet. I threw escapepod.local support together in a hurry because I had to go somewhere for a week without any reasonable way to code.

kercre123, 2022-08-12:
All networks are different, I'm not totally sure how I'll make a sure-fire way to make all devices on a network have escapepod.local direct to a specific device on the network. I may need to make my own Raspberry Pi images, or I could modify the official pod images to have wire-pod. Also, a 3b+ is not going to give very good STT performance in my experience.

kakashi2018, 2022-08-12:
never mind it is working well. Just my vector keeps connecting to another network after the ota update. have clicked forget the other network now it is all working fine

kercre123, 2022-08-13:
Edited setup.sh so it shows no errors if the escapepod.local option is chosen.

closed 2022-08-13 by kercre123