kercre123/wire-pod #5: Vector stuck in a loop three days after upgrade to wire-pod

ophidiasnaketongue, 2022-08-13:
Hi all, I've got an issue with my Vector, and it's only happened after I updated him to this firmware. He has now got a habit of getting stuck in a loop. He does the following:
-Raises his arms fully
-Drops his arms
-Thumps the ground
-Giggles

Although his back lights up on 'Hey Vector', he mishears commands and mostly produces intent-system-noaudio as a result.

He is a production model Vector running the 1.8 firmware from Project Victor. Wire-pod is being run on a Raspberry Pi 4B, 8gig, on Raspbian 64bit.

Restarting him makes no difference- it seems to have got steadily worse and he is completely stuck this morning. I have tried cleaning his wheels and infrared sensors on his feet and front. I am currently trying just leaving him to run out of battery to see if that breaks him out of his loop.

Any suggestions?

kercre123, 2022-08-13:
Are you able to manually move his head while he is off, and if you can does it make weird noises? That loop usually means the head is stuck, and he can't calibrate his IMU because of it.

ophidiasnaketongue, 2022-08-13:
Thanks for your reply- very much appreciate it! I have just checked and I can move his head while he is powered down. There are no noticeable odd noises when I do so.

kercre123, 2022-08-13:
May you send a video of him while in the loop?

ophidiasnaketongue, 2022-08-13:
I think this shows it ok:
[https://youtube.com/shorts/eYpdg7Phbq4](url)

kercre123, 2022-08-13:
Yeah, his head seems to be getting stuck near the middle. It's a hardware issue and can usually be solved by taking him apart and cleaning him out.

ophidiasnaketongue, 2022-08-13:
I guess that is my next challenge then! Thanks for your answer. I will let you know how I get on.

ophidiasnaketongue, 2022-08-13:
Further to this, I just did a partial breakdown and removed quite a lot of dust from Vector's neck.  When I switched him back on, he continued his loop until I forced his head down, at which point he reset and now _seems_ to be working correctly. Many thanks for your help, especially since this was obviously not a result of your amazing wire-pod.

I guess any further issues may be a result of a faulty IMU, but let us hope it's fixable with further cleanings!

closed 2022-08-14 by kercre123