kercre123/wire-pod #9: App support

ThommoMC, 2022-08-17:
I can’t get wire pod to work with the app. It says Vector can’t connect to the internet even though he is. Is this a side effect of using wire pod or is it because I came from the escapepod? If it is wire pod, will it ever support the Vector app?

dercodercom, 2022-08-17:
So I can use the Vector iOS app without any problems.

kercre123, 2022-08-17:
With an OSKR/Dev bot the app should work fine. But there is no app for the escape pod OTA yet, so it won't work with production robots on the escape pod OTA.

dercodercom, 2022-08-17:
> With an OSKR/Dev bot the app should work fine. But there is no app for the escape pod OTA yet, so it won't work with production robots on the escape pod OTA.

So for me the app works without problems with a production Vector.

ThommoMC, 2022-08-17:
It might be because I went straight to wire pod instead of returning to prod servers. Normally you have to clear user data, but I did not do that when migrating.

kercre123, 2022-08-18:
You were on the escape pod OTA and went directly to wire-pod? If so then that would be the reason.

ThommoMC, 2022-08-19:
Yeah that is the case. If I where to follow returning to prod servers instructions and then reset up vector with wire pod would the app begin working again.

ThommoMC, 2022-08-21:
So I reset Vector back to prod and then moved him to wire-pod and everything worked fine.

closed 2022-08-21 by ThommoMC