kercre123/wire-pod #10: Daemon unable to open libstt.so

Sovenger, 2022-08-23:
I've turned on daemon-enable and when checking the status with journalctl, I see the following error:

`Aug 22 13:16:57 escape pod start.sh[4492]: ./chipper: error while loading shared libraries: libstt.so: cannot open shared object file: No such file or directory`

Any idea what is going on? I had to install golang 1.19 and manually set up some go related exports on the path in .profile. Aside from that, I'm just running the normal command of `sudo ./setup.sh daemon-enable`.

Sovenger, 2022-08-23:
I found one hit for libstt.so in the filesystem:
`
pi@escapepod:~ $ sudo find / -name libstt.so
/root/.coqui/libstt.so
`

Are we running into an issue where this line in setup.sh isn't pointing to the expected location?
https://github.com/kercre123/wire-pod/blob/7931da38e676dc6c943bddb04610ec744a925317/setup.sh#L638

Sovenger, 2022-08-23:
I made some headway. I'm using Picovoice and it looks like the daemon doesn't build the Picovoice go file when STT_SERVICE is set to leopard. 

https://github.com/kercre123/wire-pod/blob/7931da38e676dc6c943bddb04610ec744a925317/setup.sh#L639

I believe we would need a conditional check like so:

```diff
    ...
    echo "wire-pod.service created, building chipper..."
    cd chipper
+   if [[ ${STT_SERVICE} == "leopard" ]]; then
+     /usr/local/go/bin/go run cmd-leopard/main.go
+   else
      export CGO_LDFLAGS="-L$HOME/.coqui/"
      export CGO_CXXFLAGS="-I$HOME/.coqui/"
      export LD_LIBRARY_PATH="$HOME/.coqui/:$LD_LIBRARY_PATH"
      /usr/local/go/bin/go build cmd/main.go
+   fi
     mv main chipper
     ...
```

Happy to make a pull-request. Just let me know.

KockBurn, 2022-08-24:
> I made some headway. I'm using Picovoice and it looks like the daemon doesn't build the Picovoice go file when STT_SERVICE is set to leopard.
> 

100%, I ran in to this yesterday as well.

Sovenger, 2022-08-26:
> > I made some headway. I'm using Picovoice and it looks like the daemon doesn't build the Picovoice go file when STT_SERVICE is set to leopard.
> > https://github.com/kercre123/wire-pod/blob/7931da38e676dc6c943bddb04610ec744a925317/setup.sh#L639
> > 
> > I believe we would need a conditional check like so:
> > ```diff
> >     ...
> >     echo "wire-pod.service created, building chipper..."
> >     cd chipper
> > +   if [[ ${STT_SERVICE} == "leopard" ]]; then
> > +     /usr/local/go/bin/go run cmd-leopard/main.go
> > +   else
> >       export CGO_LDFLAGS="-L$HOME/.coqui/"
> >       export CGO_CXXFLAGS="-I$HOME/.coqui/"
> >       export LD_LIBRARY_PATH="$HOME/.coqui/:$LD_LIBRARY_PATH"
> >       /usr/local/go/bin/go build cmd/main.go
> > +   fi
> >      mv main chipper
> >      ...
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Happy to make a pull-request. Just let me know.
> 
> A pull request would be appreciated. Though make sure you are getting the variable from the source.sh file

> 

I've made the changes and am testing them out before submitting a pull request. I see a few panics from Picovoice during operation though. Should I open a separate issue? Here is the output of `journalctl -fe | grep start.sh` while the daemon is running and I ask it a knowledge graph question.

```
Aug 26 19:46:32 escapepod start.sh[1153]: Initializing 3 Picovoice Instances...
Aug 26 19:46:32 escapepod start.sh[1153]: Initializing Picovoice Instance 0
Aug 26 19:46:33 escapepod start.sh[1153]: Initializing Picovoice Instance 1
Aug 26 19:46:35 escapepod start.sh[1153]: Initializing Picovoice Instance 2
Aug 26 19:46:36 escapepod start.sh[1153]: Houndify for knowledge graph initialized!
Aug 26 19:46:36 escapepod start.sh[1153]: Starting webserver at port 8080 (http://localhost:8080)
Aug 26 19:46:36 escapepod start.sh[1153]: {"function":"server.(*Server).Start","grpc-address":{"IP":"::","Port":443,"Zone":""},"http-address":null,"level":"info","msg":"server starting","source":"server.go:198","time":"2022-08-26 19:46:36"}
Aug 26 19:46:36 escapepod start.sh[1153]: Server started successfully!
Aug 26 19:46:49 escapepod start.sh[1153]: {"action":"recv_first_request","function":"server.(*Server).StreamingIntentGraph","level":"info","msg":"EOF","source":"intent_graph.go:20","status":"error","time":"2022-08-26 19:46:49"}
Aug 26 19:46:56 escapepod start.sh[1153]: {"action":"recv_first_request","function":"server.(*Server).StreamingIntentGraph","level":"info","msg":"EOF","source":"intent_graph.go:20","status":"error","time":"2022-08-26 19:46:56"}
Aug 26 19:47:04 escapepod start.sh[1153]: {"action":"recv_first_request","function":"server.(*Server).StreamingIntentGraph","level":"info","msg":"rpc error: code = DeadlineExceeded desc = context deadline exceeded","source":"intent_graph.go:20","status":"error","time":"2022-08-26 19:47:04"}
Aug 26 19:47:46 escapepod start.sh[1153]: {"action":"recv_first_request","function":"server.(*Server).StreamingIntentGraph","level":"info","msg":"rpc error: code = DeadlineExceeded desc = context deadline exceeded","source":"intent_graph.go:20","status":"error","time":"2022-08-26 19:47:46"}
Aug 26 19:52:52 escapepod start.sh[1153]: Bot 1 ESN: 00805f47
Aug 26 19:52:52 escapepod start.sh[1153]: Bot 1 Session: 74733199-4bd9-42
Aug 26 19:52:52 escapepod start.sh[1153]: Bot 1 Language: ENGLISH_US
Aug 26 19:52:52 escapepod start.sh[1153]: Stream 1 opened.
Aug 26 19:52:52 escapepod start.sh[1153]: Bot 1 Stream Type: Opus
Aug 26 19:52:52 escapepod start.sh[1153]: Processing...
Aug 26 19:52:53 escapepod start.sh[1153]: Speech completed in 0.25 seconds.
Aug 26 19:52:53 escapepod start.sh[1153]: panic: runtime error: invalid memory address or nil pointer dereference
Aug 26 19:52:53 escapepod start.sh[1153]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x8915c4]
Aug 26 19:52:53 escapepod start.sh[1153]: goroutine 83 [running]:
Aug 26 19:52:53 escapepod start.sh[1153]: github.com/Picovoice/leopard/binding/go.(*nativeLeopardType).nativeProcess(0x22?, 0x89f328?, {0x40003f0000, 0x2d00, 0x2d00})
Aug 26 19:52:53 escapepod start.sh[1153]:         /root/go/pkg/mod/github.com/!picovoice/leopard/binding/go@v1.1.1/leopard_native.go:228 +0xa4
Aug 26 19:52:53 escapepod start.sh[1153]: github.com/Picovoice/leopard/binding/go.(*Leopard).Process(0x299db08?, {0x40003f0000?, 0x40000bb2b8?, 0x1?})
Aug 26 19:52:53 escapepod start.sh[1153]:         /root/go/pkg/mod/github.com/!picovoice/leopard/binding/go@v1.1.1/leopard.go:201 +0x48
Aug 26 19:52:53 escapepod start.sh[1153]: github.com/digital-dream-labs/chipper/pkg/voice_processors/wirepod-leopard.sttHandler({0x9834c0?, 0x4000165810?}, 0x0)
Aug 26 19:52:53 escapepod start.sh[1153]:         /home/ubuntu/wire-pod/chipper/pkg/voice_processors/wirepod-leopard/stthandler.go:281 +0x155c
Aug 26 19:52:53 escapepod start.sh[1153]: github.com/digital-dream-labs/chipper/pkg/voice_processors/wirepod-leopard.(*Server).ProcessIntentGraph(0x29a4a20?, 0x4000165810)
Aug 26 19:52:53 escapepod start.sh[1153]:         /home/ubuntu/wire-pod/chipper/pkg/voice_processors/wirepod-leopard/intent_graph.go:11 +0x30
Aug 26 19:52:53 escapepod start.sh[1153]: github.com/digital-dream-labs/chipper/pkg/server.(*Server).StreamingIntentGraph(0x4000079100, {0x29a3cc8, 0x4000033d50})
Aug 26 19:52:53 escapepod start.sh[1153]:         /home/ubuntu/wire-pod/chipper/pkg/server/intent_graph.go:25 +0x1c4
[... cut ...]
Aug 26 19:52:53 escapepod start.sh[1153]:         /root/go/pkg/mod/google.golang.org/grpc@v1.48.0/server.go:1640 +0x828
Aug 26 19:52:53 escapepod start.sh[1153]: google.golang.org/grpc.(*Server).serveStreams.func1.2()
Aug 26 19:52:53 escapepod start.sh[1153]:         /root/go/pkg/mod/google.golang.org/grpc@v1.48.0/server.go:932 +0x88
Aug 26 19:52:53 escapepod start.sh[1153]: created by google.golang.org/grpc.(*Server).serveStreams.func1
Aug 26 19:52:53 escapepod start.sh[1153]:         /root/go/pkg/mod/google.golang.org/grpc@v1.48.0/server.go:930 +0x29c
```

Sovenger, 2022-08-26:
When I get lucky and Pinovoice does not panic during a knowledge graph question, I receive this parsing error. This happens for every knowledge graph request. Again, should this be a separate issue?

```
Aug 26 20:02:56 escapepod start.sh[1385]: Initializing 3 Picovoice Instances...
Aug 26 20:02:56 escapepod start.sh[1385]: Initializing Picovoice Instance 0
Aug 26 20:02:57 escapepod start.sh[1385]: Initializing Picovoice Instance 1
Aug 26 20:02:58 escapepod start.sh[1385]: Initializing Picovoice Instance 2
Aug 26 20:03:00 escapepod start.sh[1385]: Houndify for knowledge graph initialized!
Aug 26 20:03:00 escapepod start.sh[1385]: {"function":"server.(*Server).Start","grpc-address":{"IP":"::","Port":443,"Zone":""},"http-address":null,"level":"info","msg":"server starting","source":"server.go:198","time":"2022-08-26 20:03:00"}
Aug 26 20:03:00 escapepod start.sh[1385]: Server started successfully!
Aug 26 20:03:00 escapepod start.sh[1385]: Starting webserver at port 8080 (http://localhost:8080)
Aug 26 20:03:18 escapepod start.sh[1385]: Bot 1 ESN: 00805f47
Aug 26 20:03:18 escapepod start.sh[1385]: Bot 1 Session: 2de3f648-b524-43
Aug 26 20:03:18 escapepod start.sh[1385]: Bot 1 Language: ENGLISH_US
Aug 26 20:03:18 escapepod start.sh[1385]: Stream 1 opened.
Aug 26 20:03:18 escapepod start.sh[1385]: Bot 1 Stream Type: Opus
Aug 26 20:03:18 escapepod start.sh[1385]: Processing...
Aug 26 20:03:19 escapepod start.sh[1385]: Speech completed in 0.78 seconds.
Aug 26 20:03:19 escapepod start.sh[1385]: Bot 1 Transcribed text: i have a question
Aug 26 20:03:19 escapepod start.sh[1385]: Bot 1 request served.
Aug 26 20:03:19 escapepod start.sh[1385]: Bot 1 Intent Sent: intent_knowledge_promptquestion
Aug 26 20:03:19 escapepod start.sh[1385]: No Parameters Sent
Aug 26 20:03:19 escapepod start.sh[1385]: Bot 1 request served.
Aug 26 20:03:22 escapepod start.sh[1385]: Bot 1 ESN: 00805f47
Aug 26 20:03:22 escapepod start.sh[1385]: Bot 1 Session: e4e9063b-481a-4f
Aug 26 20:03:22 escapepod start.sh[1385]: Bot 1 Language: ENGLISH_US
Aug 26 20:03:22 escapepod start.sh[1385]: KG Stream 1 opened.
Aug 26 20:03:22 escapepod start.sh[1385]: Bot 1 Stream Type: Opus
Aug 26 20:03:22 escapepod start.sh[1385]: Processing...
Aug 26 20:03:23 escapepod start.sh[1385]: Speech completed in 1.38 seconds.
Aug 26 20:03:23 escapepod start.sh[1385]: Sending requst to Houndify...
Aug 26 20:03:23 escapepod start.sh[1385]: fail reading hound server message
Aug 26 20:03:23 escapepod start.sh[1385]: error response
Aug 26 20:03:23 escapepod start.sh[1385]: invalid character 'a' looking for beginning of value
Aug 26 20:03:23 escapepod start.sh[1385]: Transcribed text:
Aug 26 20:03:23 escapepod start.sh[1385]: Bot 1 request served.
Aug 26 20:03:23 escapepod start.sh[1385]: (KG) Bot 1 request served.
```

kercre123, 2022-08-28:
> When I get lucky and Picovoice does not panic during a knowledge graph question, I receive this parsing error. This happens for every knowledge graph request. Again, should this be a separate issue?

The panics seem to be because Picovoice is not receiving a long enough audio stream to process. Does it happen while you are saying the command or after you are done saying it? As for houndify, it seems you may have bad API keys.

Sovenger, 2022-08-29:
> 

> The panics seem to be because Picovoice is not receiving a long enough audio stream to process. Does it happen while you are saying the command or after you are done saying it? As for houndify, it [...]

Ah, the two must be related. I had a zero which needed to be a capitol "o" in my Client Key. Once I corrected that, both issues went away. I'll go ahead and submit this pull request now. Thanks.

PR #11 (2022-08-30): Picovoice fix for daemon-enable 
This fixes #10 by checking if Picovoice (leopard) is used in source.sh and then choosing the correct STT service to build.
closed 2022-08-30 by kercre123