kercre123/wire-pod #16: Vector won't connect to wifi

rnorganwall, 2022-10-12:
Hi! This may be a simple fix but Vector displays the "not connected" WIFI sign whenever I try to ask him a question. Wire pod server started successfully and Vector is on 1.8.1.605EP. Coqui test successful and houndify initialized. Any ideas?

kercre123, 2022-10-15:
The WiFi sign shows up when there is any issue with the voice command, so he is probably just unable to communicate with the server. Since you are using a prod Vector, I recommend using [wire-prod-pod](https://github.com/kercre123/wire-prod-pod) which has clearer instructions for setting up a custom voice server for a regular Vector.

GreudeFumat, 2022-11-16:
Hello, i'm having the same issue. it's reacting to "hi, vector", but if i ask him to do pretty much anything, from "come here" to "dance", he doesn't seem to understand, than the wifi icon starts to pop-up. Kindly assist

kercre123, 2022-11-24:
> Hello, i'm having the same issue. it's reacting to "hi, vector", but if i ask him to do pretty much anything, from "come here" to "dance", he doesn't seem to understand, than the wifi icon starts to pop-up. Kindly assist

Are you using this repo (wire-pod) or [wire-prod-pod](https://github.com/kercre123/wire-prod-pod)? wire-prod-pod has clearer instructions for getting a production bot to work.

EchoInCloud, 2022-12-03:
The included wire-pod and wire-prod-pod models are base dynamic models that need to be upgrade.. these vectors don't understand the accent either..

EchoInCloud, 2022-12-03:
The included wire-pod and wire-prod-pod models are base dynamic models that need to be upgrade.. with these models, the vectors do not even understand the accent.. etc..

kercre123, 2022-12-19:
Closing due to inactivity

closed 2022-12-19 by kercre123

nguyenhunga5, 2023-09-24:
I have the same issue, when ask for everything, the vector always shows the wifi lost signal. I can control vector from http://escapepod.local:8080