kercre123/wire-pod #19: Cannot run as service

MTrab, 2022-10-27:
Any ideas??

```
wirepod@escapepod:~ $ sudo systemctl start wire-pod
wirepod@escapepod:~ $ sudo systemctl status wire-pod
● wire-pod.service - Wire Escape Pod (coqui)
     Loaded: loaded (/lib/systemd/system/wire-pod.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2022-10-27 19:56:22 CEST; 4s ago
    Process: 731 ExecStart=/home/wirepod/wire-pod/chipper/start.sh (code=exited, status=1/FAILURE)
   Main PID: 731 (code=exited, status=1/FAILURE)
        CPU: 35ms

Oct 27 19:56:22 escapepod systemd[1]: Started Wire Escape Pod (coqui).
Oct 27 19:56:22 escapepod start.sh[734]: go: module cache not found: neither GOMODCACHE nor GOPATH is set
Oct 27 19:56:22 escapepod systemd[1]: wire-pod.service: Main process exited, code=exited, status=1/FAILURE
Oct 27 19:56:22 escapepod systemd[1]: wire-pod.service: Failed with result 'exit-code'.
```

Wire-pod is giving me (more) grey hair.
It ought to be SO easy, but haven't had much luck yet :(

```
# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye
```
Running as Raspbian on a RPI 4 4GB

kercre123, 2022-11-06:
May you send the output of `ls /home/wirepod/wire-pod/chipper`? Could you also run `./setup.sh daemon-enable` in the wire-pod directory again and paste the full output here?

MTrab, 2022-11-07:
```
$ ls /home/wirepod/wire-pod/chipper
build.sh  cmd-leopard  docker-compose.yaml  epod    go.sum   pkg        source.sh  stttest.pcm  weather-map.json
cmd       cmd-vosk     Dockerfile           go.mod  LICENSE  README.md  start.sh   useepod      webroot
```
```
$ ./setup.sh daemon-enable

Debian-based Linux confirmed.
aarch64 architecture confirmed.
This script must be run as root. sudo ./setup.sh
```
```
$ sudo ./setup.sh daemon-enable

Debian-based Linux confirmed.
aarch64 architecture confirmed.
Checks have passed!

cat: ./epod/ep.crt: No such file or directory
cat: ./epod/ep.key: No such file or directory
[Unit]
Description=Wire Escape Pod (coqui)

[Service]
Type=simple
WorkingDirectory=/home/wirepod/wire-pod/chipper
ExecStart=/home/wirepod/wire-pod/chipper/start.sh

[Install]
WantedBy=multi-user.target

wire-pod.service created, building chipper with VOSK STT service...

./chipper/chipper has been built!
Created symlink /etc/systemd/system/multi-user.target.wants/wire-pod.service → /lib/systemd/system/wire-pod.service.

systemd service has been installed and enabled! The service is called wire-pod.service

To start the service, run: 'systemctl start wire-pod'
Then, to see logs, run 'journalctl -fe | grep start.sh'
```

Now I can't even do a ´sudo systemctl start wire-pod´ doesn't exist.

MTrab, 2022-11-07:
Okay - for some reason systemctl daemon-reload wasn't run.
After running that manually it seems like it's staring this time. Perhaps this was related to the failed VOSK build in another issue

closed 2022-11-24 by kercre123