kercre123/wire-pod #22: Error building VOSK

MTrab, 2022-10-30:
Error when enabling Systemd daemon on Raspberry.

``` $ sudo ./setup.sh daemon-enable

Debian-based Linux confirmed.
aarch64 architecture confirmed.
Checks have passed!

cat: ./epod/ep.crt: No such file or directory
cat: ./epod/ep.key: No such file or directory
[Unit]
Description=Wire Escape Pod (coqui)

[Service]
Type=simple
WorkingDirectory=/home/wirepod/wire-pod/chipper
ExecStart=/home/wirepod/wire-pod/chipper/start.sh

[Install]
WantedBy=multi-user.target

wire-pod.service created, building chipper with VOSK STT service...
# github.com/alphacep/vosk-api/go
/root/go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.43/vosk.go:6:11: fatal error: vosk_api.h: No such file or directory
    6 | // #include <vosk_api.h>
      |           ^~~~~~~~~~~~
compilation terminated.
mv: cannot stat 'main': No such file or directory

./chipper/chipper has been built!
Created symlink /etc/systemd/system/multi-user.target.wants/wire-pod.service → /lib/systemd/system/wire-pod.service.

systemd service has been installed and enabled! The service is called wire-pod.service

To start the service, run: 'systemctl start wire-pod'
Then, to see logs, run 'journalctl -fe | grep start.sh'```

kercre123, 2022-11-06:
Just pushed a fix to this issue, try it now

MTrab, 2022-11-07:
That fixed this error

closed 2022-11-07 by MTrab

EchoInCloud, 2022-12-07:
Issues in setup.sh

563		echo 'export DDL_RPC_TLS_CERTIFICATE=$(cat ./epod/ep.crt)' >>source.sh
564		echo 'export DDL_RPC_TLS_KEY=$(cat ./epod/ep.key)' >>source.sh

I switch to "$(cat ./chipper/epod/..."