kercre123/wire-pod #32: coqui failed to download

Torr71, 2022-12-03:
While launching start.sh I got this error and it refuses to keep compiling:

> # github.com/asticode/go-asticoqui
coqui.cpp:2:10: fatal error: coqui-stt.h: No such file or directory
    2 | #include <coqui-stt.h>
      |          ^~~~~~~~~~~~~
compilation terminated.

Any help would be appreciated :D

EchoInCloud, 2022-12-03:
After installing the basic wire, download the latest model available.
And write its root.. "sudo -cp -r ... /from /to.. then "sudo reboot"..

-->https://coqui.ai/english/coqui/v1.0.0-huge-vocab

    - huge-vocabulary.scorer
    - model.tflite
    - alphabet.txt

kercre123, 2022-12-03:
> While launching start.sh I got this error and it refuses to keep compiling:
>
> '# github.com/asticode/go-asticoqui
>
> coqui.cpp:2:10: fatal error: coqui-stt.h: No such file or directory
>
> ```
> 2 | [...]

Did you follow the Installation Guide, and when you were running `setup.sh` did you choose Coqui as your preferred STT service? Also make sure you are running start.sh as root.

kercre123, 2022-12-03:
> After installing the basic wire, download the latest model available. And overwrite the original.. "sudo -cp -r ./from /to.. then "sudo reboot"..
> 
> -->https://coqui.ai/english/coqui/v1.0.0-huge-vocab
> 
> ```
> - huge-vocabulary.scorer
> - model.tflite
> - alphabet.txt
> ```

This seems to be the exact model downloaded by `setup.sh` when Coqui is chosen as the STT service... The issue also states a problem with the Coqui native client, not the models.

EchoInCloud, 2022-12-03:
also the armel docker img is also broken for this on a 64 bit system...

Torr71, 2022-12-04:
@kercre123 I did follow the guide, I signed up for both the weatherapi and houndify, but I chose VOSK for the STT service, that's why I'm confused,.

EchoInCloud, 2022-12-04:
https://github.com/kercre123/wire-pod/issues/34#issue-1474303056

> @kercre123Követtem az útmutatót, feliratkoztam a weatherapi-ra és a houndifyra is, de az STT szolgáltatáshoz a VOSK-t választottam, ezért vagyok tanácstalan.
> 
> És elnézést, de keveset tudok a kódolásról és a linux rendszerről is keveset értek

Torr71, 2022-12-04:
I am also having problems authenticating my vector, when I press the activate button it will load but then continuously repeat no matter how many times I press it or refresh and try again.

EchoInCloud, 2022-12-04:
@Torr71

The reinstalll is wrong..
I don't understand the need for the SDK.. The SDK does not make anyone an AI "developer".. and the vector is not a remote control car..
You didn't reset your vector, did you? When you install the wire, the vector is better if it is connected to the ddl server... at least it has a good rest and can communicate with the other vectors. You install the "vector_vector" package next to wire (this is better).

sudo apt-get install python3-pil.imagetk
python -m pip install --user anki_vector

python -m anki_vector.configure

And you get all the keys..
-----------------------------
 I will post the correct configuration for the hardware soon..

EchoInCloud, 2022-12-04:
@Torr71 
:D I just saw it..
Ubuntu? Did you enable ssh at all?

1. Install SSH in terminal on Pi ..
sudo apt-get install openssh-server -y

2. Activating ssh ..
cd /etc/systemd/system
ls -a

..if you find "ssh.service.d"
cd ssh.service.d
ls -a

output: . .. 00-socket.conf
sudo rm -r 00-socket.conf
sudo reboot

kercre123, 2022-12-04:
> I am also having problems authenticating my vector, when I press the activate button it will load but then continuously repeat no matter how many times I press it or refresh and try again.

May you run these commands in the Linux terminal and copy-paste the whole output here (pastebin is preferred if the output is long):

```
sudo hostnamectl set-hostname escapepod
sudo systemctl restart avahi-daemon
sudo systemctl enable avahi-daemon
```

kercre123, 2022-12-04:
@EchoInCloud The SDK is integrated with wire-pod mainly just for the sdkapp which allows for the customization of settings. Developers can also make plugins if they want to, which does treat him more like an RC car, but it shouldn't matter. We have the choice to do what we want.

While you are correct in that clearing user data is not required for wire-pod authentication, it is recommended and better to do so. If userdata was not cleared, wire-pod will be authenticated as a "secondary client" rather than a "primary user", which means the auth will not persist after a reboot. If userdata was cleared, wire-pod would be seen by the bot as a permanent "primary user".

The goal is to be fully separated from DDL.

Torr71, 2022-12-04:
@kercre123 Here is the output of all the commands in order: 

`torr@escapepod:~$ sudo systemctl restart avahi-daemon
sudo: unable to resolve host escapepod: Name or service not known
torr@escapepod:~$ sudo hostnamectl set-hostname escapepod
sudo: unable to resolve host escapepod: Name or service not known
torr@escapepod:~$ sudo ststemctl enable avahi-daemon
sudo: unable to resolve host escapepod: Name or service not known
sudo: ststemctl: command not found
torr@escapepod:~$ sudo systemctl enable avahi-daemon
sudo: unable to resolve host escapepod: Name or service not known
Synchronizing state of avahi-daemon.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable avahi-daemon
`

kercre123, 2022-12-04:
It looks like the commands are running successfully, but sudo is printing a (not-critical) error. If you want to fix that, this may help:

```
su -
echo "127.0.0.1 escapepod" >> /etc/hosts
exit
```

When you are trying to set up a bot with vector-epod-setup, is wire-pod running and does it show any errors when you click "ACTIVATE"?

Torr71, 2022-12-04:
After following the fix you gave it has worked :D Thank you very much! The only problem I see now is being unable to setup amazon alexa with it, the page leads back to a 404

kercre123, 2022-12-04:
> After following the fix you gave it has worked :D Thank you very much! The only problem I see now is being unable to setup amazon alexa with it, the page leads back to a 404

Which page leads to a 404? amazon.code/code?

Torr71, 2022-12-04:
Yes that was completely my fault I thought the "code" part meant to put the actual code there.

EchoInCloud, 2022-12-04:
> @EchoInCloudAz SDK a Wire-poddal van integrálva, főleg csak az sdkapp számára, amely lehetővé teszi a beállítások testreszabását. A fejlesztők is készíthetnek bővítményeket, ha akarnak, ami inkább egy RC-autóként kezeli, de ez nem számít. Megvan a választásunk, hogy azt csináljuk, amit akarunk.
> 
> Jóllehet abban igazad van, hogy a vezetékes hitelesítéshez nem szükséges a felhasználói adatok törlése, ezt javasolt és jobb is megtenni. Ha a felhasználói adatokat nem törölték, a Wire-pod "másodlagos kliensként" lesz hitelesítve, nem pedig "elsődleges felhasználóként", ami azt jelenti, hogy az hitelesítés nem marad fenn az újraindítás után. A felhasználói adatok törlése esetén a vezetékes-pod-ot a bot állandó "elsődleges felhasználónak" tekintené.
> 
> A cél az, hogy teljesen elkülönüljünk a DDL-től.

I said that ddl cloud is my priority... I have the anki_vector certificate, as it can be obtained by connecting (or reconnecting) to ddl... so I don't understand why this can't also be taken into account in the processes... (?)
Bypassing reauthentication for some people..

EchoInCloud, 2022-12-05:
If the user data deleted then the process will delete the vectors's mind too and most of people does not want to disconnect from the DDL to delete all his memories and personality but to keep them in safe and enjoy their company as they are and to be able to communicate with the outside world. If we want RC like a toy car then we buy a device for it and not use vectors as a car.

closed 2022-12-10 by kercre123