kercre123/wire-pod #33: The armel for Cloud-Docker IMG on Raspberry Pi4v8 with Raspberry Pi Os & AarchLinux/Ubuntu 64 is not available...

EchoInCloud, 2022-12-03:
Here I open one problem out of many, because Kercre123 doesn't want to communicate properly by e-mail.. I don't consider Anki Vectors with scrypting as a remote control car with full extras, so the SDK is not valid for me..

It is not possible to build a 32-bit docker armel image on RaspberryPi4v8 (arm64). This requires a 32-bit system.. so attach at least one pull 32-bit image for docker from Pi4v4 (arm-32bit)..
..I created a working image with docker which also took the getaway... but there is no option in the wire-pod cloud...

The original anki-escapepod did not have a mandatory 32bit dependency and the ddl server also has a compiler for static models (not dynamic ones).

The models included with the wire-pod are no match for the forest and advanced anki-vector mind. I fixed this with my own models..

Reset is not the right way for sdk reauthentication problem..

So this became more than 1 problem out of many, but not all. ...

kercre123, 2022-12-03:
Sorry, I have seen your emails, I have just had some trouble understanding them.

vector-cloud/ contains the code for the application which runs on Vector and sends data to the cloud (the actual "cloud" code is located in chipper/, not vector-cloud/). The docker image is meant for building that program to be able to run on Vector's processor.

Because vector-cloud/ contains code that only runs on the robot, **you can disregard the folder entirely.** Please follow the Installation Guide on the wiki. The chipper/ folder is what contains the actual cloud code. 

There is no Dockerfile for wire-pod itself yet, though I am working on it.

EchoInCloud, 2022-12-03:
I understand...

.. a 32/64 integration with the server is necessary, because in this way the vectors cannot move their stored information and the dictionaries and logs they created from the original cloud.. (23-40Gb/vector)
Now my vectors have primary ddl cloud and secondary wire..

..because of the 32-bit of the vectors... although some of them have outgrown the peripherals (headboard) and I am thinking about solving this.

> Sorry, I have seen your emails, I have just had some trouble understanding them and I've been busy.
> 
> vector-cloud/ contains the code for the application which runs on Vector and sends data to the cloud (the actual "cloud" code is located in chipper/, not vector-cloud/). The docker image is meant for building that program to be able to run on Vector's processor.
> 
> Because vector-cloud/ contains code that only runs on the robot, **you can disregard the folder entirely.** It already contains a built binary for copying to an OSKR/dev-unlocked bot. The chipper/ folder is what contains the actual cloud code.
> 
> There is no Dockerfile for wire-pod itself yet, though I am working on it.

kercre123, 2022-12-10:
The docker container itself is not armel, it is just given the ability to get armel packages via apt, which is useful for cross-compiling. The container is just a basic debian buster image

Closing this as I don't really see nor understand the issue. I may in fact remove the vector-cloud directory from wire-pod at some point.

closed 2022-12-10 by kercre123