kercre123/wire-pod #34: Replace the vosk model ...

EchoInCloud, 2022-12-04:
Tested... it's worth replacing the vosk-model-small-en-us-0.15 in the wire --> https://alphacephei.com/vosk/models vosk-model-en-us-0.22-lgraph

EchoInCloud, 2022-12-04:
For already installed Wire:

Download model and extract  in "/home/pi/Downloads" ...

..from pi user.. 
sudo cp -r ./Downloads/vosk-model-en-us-0.22-lgraph/am /home/pi/wire-pod or wire-prod-pod/vosk/models/en-US/model/
sudo cp -r ./Downloads/vosk-model-en-us-0.22-lgraph/conf /home/pi/wire-pod or wire-prod-pod/vosk/models/en-US/model/
sudo cp -r ./Downloads/vosk-model-en-us-0.22-lgraph/graph /home/pi/wire-pod or wire-prod-pod/vosk/models/en-US/model/
sudo cp -r ./Downloads/vosk-model-en-us-0.22-lgraph/ivector /home/pi/wire-pod or wire-prod-pod/vosk/models/en-US/model/

sudo reboot

kercre123, 2022-12-04:
Tried it out, the performance on some of my hardware is not great with it so I will not make the 0.22 model default. I may add it as an option though.

closed 2022-12-04 by kercre123