kercre123/wire-pod #35: Error launching chipper

correderadiego, 2022-12-04:
Good afternoon ,

First of all I would like to thank you all your effort with this project. It is awesome.

After all of it , I am having a little issue with it and I would like to know if you could help me. I am trying to install wire-pod in a linux mint laptop. I run the wire-pod setup.sh and I got problems launching the chipper server. I add all the command line information : 

**sudo ./setup.sh**
Debian-based Linux confirmed.
amd64 architecture confirmed.
AVX support confirmed.
Checks have passed!

What would you like to do?
1: Full Setup (recommended) (builds chipper, gets STT stuff, generates certs, creates source.sh file, and creates server_config.json for your bot
2: Just build vic-cloud
3: Just build chipper
4: Just get STT assets
5: Just generate certs
6: Create wire-pod config file (change/add API keys)
(NOTE: You can just press enter without entering a number to select the default, recommended option)

Enter a number (1): 4

Which speech-to-text service would you like to use?
1: Coqui (local, no usage collection, less accurate, a little slower)
2: Picovoice Leopard (local, usage collected, accurate, account signup required)
3: VOSK (local, accurate, multilanguage, fast)

Enter a number (3):  

Which STT language would you like to use?
1: English (US)
2: Italian (IT)

Enter a number (1): 1

**sudo ./chipper/start.sh**
pkg-config --cflags  -- opus opus opus opus opusfile opusfile
Package opus was not found in the pkg-config search path.
Perhaps you should add the directory containing `opus.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opus' found
Package opus was not found in the pkg-config search path.
Perhaps you should add the directory containing `opus.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opus' found
Package opus was not found in the pkg-config search path.
Perhaps you should add the directory containing `opus.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opus' found
Package opus was not found in the pkg-config search path.
Perhaps you should add the directory containing `opus.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opus' found
Package opusfile was not found in the pkg-config search path.
Perhaps you should add the directory containing `opusfile.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opusfile' found
Package opusfile was not found in the pkg-config search path.
Perhaps you should add the directory containing `opusfile.pc'
to the PKG_CONFIG_PATH environment variable
No package 'opusfile' found
pkg-config: exit status 1
github.com/alphacep/vosk-api/go
_/root/go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.43/vosk.go:6:11: fatal error: vosk_api.h: No such file or directory
    6 | // #include <vosk_api.h>
      |           ^~~~~~~~~~~~_
compilation terminated.

EchoInCloud, 2022-12-04:
ohh yes..
I didn't mention the opus bug before.. the opus linker in the wire package is faulty.. I built opus from source.. simple and fast, smooth build (this is the best alternative).. but @kercre123  should provide the opportunity to refer to the opus built on the server..

Log:
... ... ...
pi@escapepod:~/wire-prod-pod/vector-cloud $ cd ../../opus
pi@escapepod:~/opus $ ./autogen.sh

Updating build configuration files, please wait....
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:38: installing './compile'
configure.ac:36: installing './config.guess'
configure.ac:36: installing './config.sub'
configure.ac:33: installing './install-sh'
configure.ac:33: installing './missing'
Makefile.am:347: warning: '%'-style pattern rules are a GNU make extension
Makefile.am:350: warning: '%'-style pattern rules are a GNU make extension
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'

pi@escapepod:~/opus $ ./configure

checking whether make supports nested variables... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
...
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
...
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
...
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
...
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
[... 164 lines cut ...]
PASS: celt/tests/test_unit_rotation
PASS: celt/tests/test_unit_types
PASS: silk/tests/test_unit_LPC_inv_pred_gain
PASS: tests/test_opus_api
PASS: tests/test_opus_decode
PASS: tests/test_opus_encode
PASS: tests/test_opus_padding
PASS: tests/test_opus_projection
============================================================================
Testsuite summary for opus 1.3.1-158-gbce1f392
============================================================================
# TOTAL: 14
# PASS:  14
# SKIP:  0
# XFAIL: 0
# FAIL:  0
# XPASS: 0
pi@escapepod:~/opus $
============================================================================
make[4]: Leaving directory '/home/pi/opus'
make[3]: Leaving directory '/home/pi/opus'
make[2]: Leaving directory '/home/pi/opus'
make[1]: Leaving directory '/home/pi/opus'

pi@escapepod:~/opus $

EchoInCloud, 2022-12-04:
ohh yes..
I didn't mention the opus bug before.. the opus linker in the wire package is faulty.. I built opus from source.. simple and fast, smooth build (this is the best alternative).. but @kercre123   should provide the opportunity to refer to the opus built on the server..

Log:
... ... ...
pi@escapepod:~/wire-prod-pod/vector-cloud $ cd ../../opus
pi@escapepod:~/opus $ ./autogen.sh

Updating build configuration files, please wait....
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:38: installing './compile'
configure.ac:36: installing './config.guess'
configure.ac:36: installing './config.sub'
configure.ac:33: installing './install-sh'
configure.ac:33: installing './missing'
Makefile.am:347: warning: '%'-style pattern rules are a GNU make extension
Makefile.am:350: warning: '%'-style pattern rules are a GNU make extension
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'

pi@escapepod:~/opus $ ./configure

checking whether make supports nested variables... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
... ... ...
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
. . .
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
. . .
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
. . .
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
[... 162 lines cut ...]
PASS: celt/tests/test_unit_rotation
PASS: celt/tests/test_unit_types
PASS: silk/tests/test_unit_LPC_inv_pred_gain
PASS: tests/test_opus_api
PASS: tests/test_opus_decode
PASS: tests/test_opus_encode
PASS: tests/test_opus_padding
PASS: tests/test_opus_projection
= = =
Testsuite summary for opus 1.3.1-158-gbce1f392
= = =
TOTAL: 14
PASS:  14
SKIP:  0
XFAIL: 0
FAIL:  0
XPASS: 0
pi@escapepod:~/opus $
= = =
make[4]: Leaving directory '/home/pi/opus'
make[3]: Leaving directory '/home/pi/opus'
make[2]: Leaving directory '/home/pi/opus'
make[1]: Leaving directory '/home/pi/opus'

pi@escapepod:~/opus $

EchoInCloud, 2022-12-04:
ohh yes..
I didn't mention the opus bug before.. the opus linker in the wire package is faulty.. I built opus from source.. simple and fast, smooth build (this is the best alternative).. but @kercre123    should provide the opportunity to refer to the opus built on the server..

Log:
... ... ...
pi@escapepod:~$ cd .../opus
pi@escapepod:~/opus $ ./autogen.sh

Updating build configuration files, please wait....
libtoolize: putting auxiliary files in '.'.
libtoolize: linking file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: linking file 'm4/libtool.m4'
libtoolize: linking file 'm4/ltoptions.m4'
libtoolize: linking file 'm4/ltsugar.m4'
libtoolize: linking file 'm4/ltversion.m4'
libtoolize: linking file 'm4/lt~obsolete.m4'
configure.ac:38: installing './compile'
configure.ac:36: installing './config.guess'
configure.ac:36: installing './config.sub'
configure.ac:33: installing './install-sh'
configure.ac:33: installing './missing'
Makefile.am:347: warning: '%'-style pattern rules are a GNU make extension
Makefile.am:350: warning: '%'-style pattern rules are a GNU make extension
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'

pi@escapepod:~/opus $ ./configure

checking whether make supports nested variables... yes
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
... ... ...
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
. . .
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
. . .
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
. . .
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
[... 162 lines cut ...]
PASS: celt/tests/test_unit_rotation
PASS: celt/tests/test_unit_types
PASS: silk/tests/test_unit_LPC_inv_pred_gain
PASS: tests/test_opus_api
PASS: tests/test_opus_decode
PASS: tests/test_opus_encode
PASS: tests/test_opus_padding
PASS: tests/test_opus_projection
= = =
Testsuite summary for opus 1.3.1-158-gbce1f392
= = =
TOTAL: 14
PASS:  14
SKIP:  0
XFAIL: 0
FAIL:  0
XPASS: 0
pi@escapepod:~/opus $
= = =
make[4]: Leaving directory '/home/pi/opus'
make[3]: Leaving directory '/home/pi/opus'
make[2]: Leaving directory '/home/pi/opus'
make[1]: Leaving directory '/home/pi/opus'

pi@escapepod:~/opus $

kercre123, 2022-12-04:
> Good afternoon ,
>
> First of all I would like to thank you all your effort with this project. It is awesome.
>
> After all of it , I am having a little issue with it and I would like to know if you could [...]

It looks like Opus is not installed, and it can be installed if you choose the 1st option in setup.sh. It seems you only chose the 4th option.

kercre123, 2022-12-04:
@EchoInCloud Please don't paste the entire output of compile commands. Either don't include it at all or use pastebin links.

EchoInCloud, 2022-12-04:
@kercre123  Okay.. but it would be good to integrate the solution I wrote with an "if"..

kercre123, 2022-12-04:
> @kercre123 Okay.. but it would be good to integrate the solution I wrote with an "if"..

I would like to see more evidence of broken libopus before integrating a build for it in setup.sh. The actual solution for this specific issue seems to be a lot more simple.

correderadiego, 2022-12-04:
> It looks like Opus is not installed, and it can be installed if you choose the 1st option in setup.sh. It seems you only chose the 4th option.

I have already used the first option and It looks like all the files has been installed 
 **sudo ./setup.sh**         

Debian-based Linux confirmed.
amd64 architecture confirmed.
AVX support confirmed.
Checks have passed!

What would you like to do?
1: Full Setup (recommended) (builds chipper, gets STT stuff, generates certs, creates source.sh file, and creates server_config.json for your bot
2: Just build vic-cloud
3: Just build chipper
4: Just get STT assets
5: Just generate certs
6: Create wire-pod config file (change/add API keys)
(NOTE: You can just press enter without entering a number to select the default, recommended option)

Enter a number (1): 1

**Required packages already gotten.**

Which speech-to-text service would you like to use?
1: Coqui (local, no usage collection, less accurate, a little slower)
2: Picovoice Leopard (local, usage collected, accurate, account signup required)
3: VOSK (local, accurate, multilanguage, fast)

Enter a number (3):

EchoInCloud, 2022-12-04:
> > @kercre123oké.. de jó lenne az általam írt megoldást integrálni egy "ha"..
> 
> Szeretnék több bizonyítékot látni a törött libopusra, mielőtt egy buildet integrálnék a setup.sh-ba. A konkrét probléma megoldása sokkal egyszerűbbnek tűnik.

it is only necessary to check the presence of opus in home/user.. if it is found, build with it..

I deleted the logo.. with the opus downloaded by wire, it didn't build on my Raspberry os64 even if the necessary dev files were installed.. these files are also needed to build the opus from source, which was successfully built.. (you deleted it)

EchoInCloud, 2022-12-04:
with "pkg-config libopus-dev libopusfile-dev"

build.sh --> cloud/main.go
. . .
command-line-arguments
/usr/local/go/pkg/tool/linux_arm64/link: internal linking requested but external linking required: some packages could not be built to support internal linking ([gopkg.in/hraban/opus.v2])
make: *** [Makefile:13: vic-cloud] Error 2

correderadiego, 2022-12-04:
> with "pkg-config libopus-dev libopusfile-dev"
> 
> build.sh --> . . . -o build/vic-cloud cloud/main.go command-line-arguments /usr/local/go/pkg/tool/linux_arm64/link: internal linking requested but external linking required: some packages could not be built to support internal linking ([gopkg.in/hraban/opus.v2]) make: *** [Makefile:13: vic-cloud] Error 2

_The simple "sudo apt-get install -y libopus-dev etc.."_ This was enough for me.

I have a last error 

# github.com/alphacep/vosk-api/go
/root/go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.43/vosk.go:6:11: fatal error: vosk_api.h: No such file or directory
    6 | // #include <vosk_api.h>
      |           ^~~~~

Any ideas?

kercre123, 2022-12-04:
Try running the following commands:

```
cd ~/wire-pod
sudo rm -rf vosk
sudo rm -f vector-cloud/packagesGotten
```

Then run setup.sh again with the 1st option and send the output here (preferably in a pastebin)

correderadiego, 2022-12-04:
I have already made it and the output si https://pastebin.com/TRHWX0K9

I have never used pastebin before so please tell me if this link is was you were expecting. 

Thanks.

kercre123, 2022-12-04:
Looks like wget is missing. Run: `sudo apt install -y wget` then re-run the commands I put in my last response, then try setup.sh with the 1st option again.

EchoInCloud, 2022-12-04:
@correderadiego 
I'm glad I could help with this..

kercre123, 2022-12-04:
> I have already made it and the output si https://pastebin.com/TRHWX0K9
> 
> I have never used pastebin before so please tell me if this link is was you were expecting.
> 
> Thanks.

Is your device's date and time set correctly? It is trying to download assets from the VOSK site but it can't verify the cert. I'll make a quick update to setup.sh which allows it to download assets insecurely.

correderadiego, 2022-12-04:
> Is your device's date and time set correctly? It is trying to download assets from the VOSK site but it can't verify the cert. I'll make a quick update to setup.sh which allows it to download assets [...]

Yes it is :)

kercre123, 2022-12-04:
Just pushed a fix, run the following:

```
cd ~/wire-pod
sudo ./update.sh
sudo rm -rf vosk
sudo rm -f vector-cloud/packagesGotten
```

Then try setup.sh again

correderadiego, 2022-12-04:
Solved, thank you so much!!!

closed 2022-12-04 by correderadiego