kercre123/wire-pod #36: sdkapp | failed: INI file missing

Falroi, 2022-12-05:
browser sdkapp isn't reading the sdk_config.ini file in /.anki_vector

Skybird129, 2022-12-05:
I've the same problem using wire-prod-pod as well.

Falroi, 2022-12-05:
Thanks, the install docs suggested otherwise. I'll try that.

Falroi, 2022-12-09:
> /.anki_vector/sdk_config.ini ?
> 
> I wrote a suggestion for scanning from here.. the file is currently marked as scanning the "getaway" build folder.. if I understood correctly...

This didn't seem to work, any other suggestions?

kercre123, 2022-12-10:
This issue should be solved now, I accidentally only had it attempt to read the INI file without trying to read the JSON it should have been reading. Try stopping wire-pod, updating it with `cd ~/wire-pod` then `sudo ./update.sh`, then start wire-pod again.

Falroi, 2022-12-10:
> This issue should be solved now, I accidentally only had it attempt to read the INI file without trying to read the JSON it should have been reading. Try stopping wire-pod, updating it with `cd ~/wire-pod` then `sudo ./update.sh`, then start wire-pod again.

Same error. The exact error is "failed: INI file missing"

kercre123, 2022-12-18:
May you try reinstalling wire-pod? That error should not exist anymore.

kercre123, 2023-01-03:
Closing due to inactivity and because this issue shouldn't occur anymore.

closed 2023-01-03 by kercre123