kercre123/wire-pod #44: Chipper Issue

mdinolfo123, 2022-12-10:
When I try to start Chipper, I get:

pkg/sdkapp/server.go:19:2: missing go.sum entry for module providing package github.com/fforchino/vector-go-sdk/pkg/vector (imported by github.com/digital-dream-labs/chipper/pkg/sdkapp); to add:
	go get github.com/digital-dream-labs/chipper/pkg/sdkapp
pkg/sdkapp/server.go:20:2: missing go.sum entry for module providing package github.com/fforchino/vector-go-sdk/pkg/vectorpb (imported by github.com/digital-dream-labs/chipper/pkg/sdkapp); to add:
	go get github.com/digital-dream-labs/chipper/pkg/sdkapp

kercre123, 2022-12-10:
Fixed now, cd into the wire-pod directory and run `sudo ./update.sh`

closed 2022-12-10 by kercre123

mdinolfo123, 2022-12-10:
Thank you so much!