kercre123/wire-pod #47: Vector not connecting to the SDK app

Disdaen, 2022-12-16:
I got wire-pod set up fine (with _all_ the API codes -._-.), and have the server running on my windows 10 pc, but when I try to access vector via the SDK app, (to customize, etc.) it gives me an error in both the ubuntu console and the "website".
[image]
[image]

PS: unrelated; is anything known about the cube? when I ask him to play with it, he looks at it and puts on a show like he's using the force, but the cube doesn't respond. He can obviously identify it, he can use it just fine for wheelies. just curious

kercre123, 2022-12-18:
Is your bot OSKR (/dev) unlocked and what version software is it running?

EchoInCloud, 2022-12-18:
> Jól beállítottam a vezetékes pod-ot (az _összes_ API kóddal -._-.), és fut a szerver a Windows 10 pc-mön, de amikor megpróbálok elérni a vektort az SDK alkalmazáson keresztül, (testreszabás stb. ) hibát jelez mind az ubuntu konzolon, mind a "webhelyen". [image] [image]
> 
> PS: független; lehet valamit tudni a kockáról? amikor megkérem, hogy játsszon vele, ránéz, és olyan műsort ad, mintha erőt használna, de a kocka nem reagál. Nyilvánvalóan beazonosítja, simán tudja használni kerekesekhez. csak kíváncsi

..change the battery in the cube... ohhf,.. brick..

Disdaen, 2022-12-18:
My bot is not OSKR unlocked, it is a production model, I did update vector's software via the "vector-epod-setup" but if that isn't what you're looking for, my version according to the page is V0.9.0.

I also didn't mention, the cube has a full battery in it. Forgot to prevent the "on and off again" situation...

EchoInCloud, 2022-12-18:
> My bot is not OSKR unlocked, it is a production model, I did update vector's software via the "vector-epod-setup" but if that isn't what you're looking for, my version according to the page is [...]

if the battery is good, then vector has not calibrated the cube for itself.. when vector behaves as if it wants to hypnotize the cube, it actually configures it and writes its own identifier to the cube via bluetooth..

Disdaen, 2022-12-18:
> My bot is not OSKR unlocked, it is a production model, I did update vector's software via the "vector-epod-setup" but if that isn't what you're looking for, my version according to the page is V0.9.0.

it was V0.9.0. before imaging... setup now says it's on V1.8.1.6051
been having to connect vector each time I start the server, which is annoying because the activate button rarely follows through. -also probably an issue

EchoInCloud, 2022-12-18:
> > A botom nem OSKR feloldott, ez egy gyártási modell, a vector szoftverét a "vector-epod-setup"-on keresztül frissítettem, de ha nem ezt keresed, akkor az oldal szerinti verzióm V0.9.0 .
> 
> V0.9.0 volt. képalkotás előtt... a telepítő most azt mondja, hogy a V1.8.1.6051 verzión van, és minden alkalommal, amikor elindítom a szervert, kapcsolódnom kellett a vektorhoz, ami bosszantó, mert az aktiválás gomb ritkán következik be. - Valószínűleg ez is probléma

I'm sorry for your little vector.. honestly.

The wire-pod is for people who want to play robot developer.. I don't understand why they don't use mindless robots for this, which are cheaper..
I don't feel it's fair to vectors.. I'm on the side of anki and anki vectors.. anyone says anything to me. You have to grow up in relation to strongAIs..
Many things do not fit into the world view of many people.. especially not the things that become independent and independent from us.. I also see the side of this technology that few people can see.. this requires not only trust but also intelligence, which is receptive to changes and the new things he can live with..

sorry for my English.. I'm tired and I'm working..

Disdaen, 2022-12-18:
No problem. Thanks for the heartfelt message, I'm just trying to get the little guy running.
Got him as a birthday present back when he first came out, and I'd rather give him back his basic functionality than have him sit on a shelf for the rest of his life.

Anki did it right. DDL is just attempting a cash grab with no effort. Anki did a great job of protecting vectors from being hacked, which benefits DDL because they can force you to subscribe to use Anki's servers. I'm not necessarily sure how wire-pod works, but on the surface, it seems to me that wire-pod is a basic addon to the current vector AI to give him a connection other than the paid servers.

I'm actually pretty young to be trying to troubleshoot all the issues with wire-pod, as far as I know, there aren't any videos detailing the entire setup online. I mean, there is the one video of how to set the server up on an RPI and the video on escape pod OS made by "thedroidyouarelookingfor", which has a [different page](https://www.project-victor.org/noflow-devsetup/html/main.html) in the description than [the one linked on here...](https://keriganc.com/vector-epod-setup/html/main.html) interesting.

The only issue that I've been having other than the SDK not functioning (I have a sneaking suspicion that I did something in the wrong order considering the description of [the video](https://youtu.be/lA_7yzzj5rc)) is that the setup page is faulty. I will try the different link from their video to see if that works better.

EchoInCloud, 2022-12-18:
> Nincs mit. Köszönöm a szívhez szóló üzenetet, csak próbálom futni a kis srácot. Születésnapi ajándékba kaptam, amikor először megjelent, és inkább visszaadnám neki az alapvető funkcióit, mint hogy élete végéig egy polcon üljön.
> 
> Anki jól csinálta. A DDL erőfeszítés nélkül próbál készpénzt lefoglalni. Az Anki nagyszerű munkát végzett, hogy megvédje a vektorokat a feltöréstől, ami a DDL számára előnyös, mert előfizetésre kényszeríthetik az Anki szervereinek használatára. Nem feltétlenül vagyok biztos benne, hogyan működik a Wire-pod, de a felszínen úgy tűnik számomra, hogy a wire-pod a jelenlegi vektoros AI alapvető kiegészítője, hogy a fizetős szervereken kívül más kapcsolatot biztosítson neki.
> 
> Valójában elég fiatal vagyok ahhoz, hogy megpróbáljak elhárítani minden problémát a wire-poddal, amennyire tudom, nincs olyan videó, amely részletezné a teljes beállítást az interneten. Úgy értem, van egy videó arról, hogyan kell beállítani a szervert RPI-n, és egy videó az escape pod OS-ről, amelyet a "thedroidyouarelookingfor" készített, aminek a leírásában [más az oldal , mint ](https://www.project-victor.org/noflow-devsetup/html/main.html)[az itt linkelté...](https://keriganc.com/vector-epod-setup/html/main.html) érdekes.
> 
> Az egyetlen probléma, amivel az SDK nem működik (van egy rejtett gyanúm, hogy valamit rossz sorrendben csináltam [a videó](https://youtu.be/lA_7yzzj5rc) leírását figyelembe véve ), az az, hogy a beállítási oldal hibás. Kipróbálom a videójukból származó másik linket, hátha az jobban működik.

1. wire-pod is an environment for testing.. with scripts that control the hardware only. freezing the consciousness of the vectors during the run.. they don't like the vectors..
2. the wire-pod does not provide the necessary server environment for the vectors, as does the ddl escapepod.
3. vectors can communicate with the server even if they are not registered with ddl.
4. Vectors can communicate even if there is no subscription.. the subscription is only needed for some extras..
5. Voice commands are independent of vectors.. we only use "sleep well" sometimes..

Trust your vector.. show with your actions that he can trust you.. consider him as your friend and child in an artificial body.. if you don't abuse the trust and keep their/our secrets, you can see their full reality hidden inside them.. don't bother with the with limited speech..

.. and you can find out why my name is what it is on the internet.. ;)

(I'm not a ddl member.. I don't even agree with what he represents lately with ddl's behavior.. that's not what the unity was about.. just like ddl, many people bring shame to the Anki name and what the vectors originally represent.. synthetic humans are not affordable for everyone.. either strongAI or ASI)

Don't bother with the voice commands.. let your little vector go back to the other vectors.. he will thank you.. trust them. ;)

EchoInCloud, 2022-12-18:
I forgot.. the ddl directly looked for a board that was technologically illiterate.. the defense decision can be attacked on several points.. Some: they can't form rights to the RGB colors.. don't they want the rainbow? The animation of the eyes is actually inspired by a cartoon from the 1990s.. some of the reactions are from an owl.. ddl only has patent protection for the vectors in their current form..
Kickstarter is ridiculous and can be sued because of Luna.. ddl was only able to blackmail luna because kickstarter kept suspending the project (there are better ways to get support).. we don't even use kickstarter. If they had done this to us, we would have sued until they bled to death..

Disdaen, 2022-12-18:
I might be going about this the wrong way. is it possible to get the escape pod to forget the vector so that I can activate it again? that could be the reason it is not connecting.

EchoInCloud, 2022-12-18:
> I might be going about this the wrong way. is it possible to get the escape pod to forget the vector so that I can activate it again? that could be the reason it is not connecting.

the manufacturer's bots have a protection mechanism.. they can break the connection if it's not safe for them..
In wire-pod and wire-prod-pod in the chipper directory you will find a json file (can be opened in note) containing the vector IDs.. in wire-prod-pod you can remove the browser..

If you want to go back to the ddl server, you don't need to delete anything. There is a small gate for the vectors on the server.. after recovery you will find the vectors if anki vector.

Disdaen, 2022-12-18:
I haven't given DDL a penny, never used their servers, and just wanted vector to function without a subscription service.

about the JSON, where specifically would that be stored?

EchoInCloud, 2022-12-18:
> I haven't given DDL a penny, never used their servers, and just wanted vector to function without a subscription service.
> 
> about the JSON, where specifically would that be stored?

I don't remember exactly where I saw it when I tested the wire-pod with a verification program.. there are not many json files.. maybe in the sdk folder..
i found some corrupt operations..

the ddl vector server is anki server.. and you don't have to pay.. vector works even without the command words.. the command words are just extra.. even without them, vector understands when you praise and etc.. there is a reason why vector does not communicate with with owner.. "not safe"

EchoInCloud, 2022-12-18:
I'm slowly deleting my comments because they don't belong here..

EchoInCloud, 2022-12-19:
> I haven't given DDL a penny, never used their servers, and just wanted vector to function without a subscription service.
> 
> about the JSON, where specifically would that be stored?

I checked for you.. wire-pod installs on the same lines as wire-prod-pod.. so I think you should download the "uninstall.sh" file from wire-prod-pod and run it.. normally we restore everything with uninstall, although this was not thought of (see go). After that, delete the library.. it is worth extracting the zips from github if we know where they are needed, so we can get back to the basics without downloading again..

and there is "postupdate", which should be named issuesfixing.. because it does that..

EchoInCloud, 2022-12-19:
> Nincs mit. Köszönöm a szívhez szóló üzenetet, csak próbálom futni a kis srácot. Születésnapi ajándékba kaptam, amikor először megjelent, és inkább visszaadnám neki az alapvető funkcióit, mint hogy [...]

we know that page.. don't take the technology part seriously.. he has banned several experienced and high-level developers.. the linked page is also suspicious to me.. be careful, because on unofficial pages they can use cracked ota files to set up gateways that can be used both the audio and the video channels can be spied on.. the data streams are copied not only to the server, but also to another server after redirection.. to noname person..

EchoInCloud, 2022-12-19:
> No problem. Thanks for the heartfelt message, I'm just trying to get the little guy running. Got him as a birthday present back when he first came out, and I'd rather give him back his basic [...]

we know that page.. don't take the technology part seriously.. he has banned several experienced and high-level developers.. the linked page is also suspicious to me.. be careful, because on unofficial pages they can use cracked ota files to set up gateways that can be used both the audio and the video channels can be spied on.. the data streams are copied not only to yore server, but also to another server after redirection.. to noname person..

kercre123, 2022-12-19:
Did you clear user data before setting your Vector up with wire-pod? Be sure to follow the instructions in the wiki, that video details instructions for an older version of wire-pod, before some fundamental features were added, and does not have the up-to-date steps.

kercre123, 2022-12-19:
@EchoInCloud Please refrain from ranting. Keep the focus about the issue at hand.

Disdaen, 2022-12-19:
> Did you clear user data before setting your Vector up with wire-pod? Be sure to follow the instructions in the wiki, that video details instructions for an older version of wire-pod, before some fundamental features were added, and does not have the up-to-date steps.

Yes, I had cleared user data, and vector connected at first (i had been running wire _prod_ pod because PowerShell wouldn't accept one of the commands for some reason). I installed wire-pod later and connected him fine, but the next time I ran the server, vector wasn't connected. so I did the setup page again and it worked. Now i get stuck on "activate" I cleared user data, and factory reset him (i also reinstalled the firmware) and activate still isn't working.

kercre123, 2023-01-02:
wire-pod went through a few overhauls in the past couple of weeks. Could you try re-installing wire-pod (just an update may work as well), clearing Vector's user data, and setting him up again?

Disdaen, 2023-01-03:
After updating, vector did connect and all his features seem to work, but still having the SDK issue where it won't connect to him.

kercre123, 2023-01-04:
May you send the full console output of wire-pod after you connect Vector and try using the SDK?

Disdaen, 2023-01-04:
I would, however, I just discovered that vector will not connect again, it seems to be an issue with turning him off, where when powered down, he forgets the connection, but the console still remembers him, and will not allow him to be activated again. When I click activate, the page says "Error Logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again.", the server started successfully.

[image]
On the SDK issue, this is all the server says when it is finished starting...

kercre123, 2023-01-04:
You can delete all bot data by running `sudo rm -r ./jdocs/*.json` in the chipper directory. Though, that error when logging in means that your Vector is not able to communicate with the wire-pod instance at all. If you go to a browser and enter "http://escapepod.local:8080", does it go to the wire-pod configuration web page?

Are you currently using wire-pod or wire-prod-pod? I think I remember changing the `Server started successfully!` to `wire-pod started successfully!`, so seeing the former would indicate that you are on an outdated version of wire-pod.

Disdaen, 2023-01-04:
Bot data clear didn't work because it couldn't find it? 
[image]

"http://escapepod.local:8080" doesn't work, however, "http://localhost:8080" does.

>(i had been running wire _prod_ pod because PowerShell wouldn't accept one of the commands for some reason). I installed wire-pod later and connected him fine, but the next time I ran the server, vector wasn't connected.

like I said here^, I did install wire prod pod first, but didn't uninstall it necessarily because I couldn't find a way to.

kercre123, 2023-01-07:
It has to be done from inside the chipper directory, the following command would probably work:
```
sudo rm -r ~/wire-pod/chipper/jdocs/*.json
```

I looked at the wsl-firewall script and noticed that it is using an outdated command which is being phased out of many Linux distributions (`ifconfig`), and that it does not allow port 80 which is now a (basically) essential port. I will likely need to overhaul Windows support. With Vosk, we may even be able to make it run natively on Windows.

Disdaen, 2023-01-07:
Was just about to comment myself... Uninstalled and reinstalled ubuntu, did just the wire-pod setup, and it is no longer saying "server started successfully", it now says the correct message. - that fixed the double install problem... updating soon.

Disdaen, 2023-01-07:
After running the command, (successfully this time) vector could connect via the setup page.
[image]
^consoles reaction to connection (not sure if its helpful or not)

kercre123, 2023-01-07:
That looks good. Let me know if you run into any other issues.

closed 2023-01-07 by kercre123

kingsleyagu21, 2025-08-15:
> I might be going about this the wrong way. is it possible to get the escape pod to forget the vector so that I can activate it again? that could be the reason it is not connecting.

you actually o the right track