kercre123/wire-pod #55: I got pinged in a random discussion post... don't know why, but kinda want it removed ngl

sparty182020, 2022-12-29:
the comment contains "nsfw" (not gonna click the link as its prob a scam), but there were others also pinged so I was hopping it could get removed and/or reported...

Link: https://github.com/kercre123/wire-pod/discussions/24#discussioncomment-4519164

sparty182020, 2022-12-29:
also looking at the top languages in this repo, I have no experience in the top 2, but have some Shell experience and I know a lot of HTML, CSS, and Javascript so if you need any help with that, lmk

kercre123, 2022-12-29:
A couple of those comments were made, they have been reported and the associated posts have been deleted. Sorry about the ping.

closed 2022-12-29 by kercre123

sparty182020, 2022-12-29:
Its all cool man. Like i said, I can also help with any HTML, JavaScript, or CSS code if it isn't working as I'm good at those languages