kercre123/wire-pod #76: most of the settings don't work anymore

Hamlet3000, 2023-02-03:
The last built seems broken. Most of the settings don't work anymore like Volume, Locale, Temperature Units, Button Action, Set Location, and Set Time Zone.

I have completely reinstalled your original branch.

kercre123, 2023-02-03:
Try going to :8080/sdkapp and running Ctrl+Shift+R (CMD+Shift+R on Mac) on the main webpage to redownload all of the scripts and such, they may be caching

Hamlet3000, 2023-02-03:
I have already done that. I have completely re-downloaded the sources and gone through the full istallation process. I have also tried different browsers on different machines.

kercre123, 2023-02-03:
Do any errors appear in the wire-pod logs or the browser log?

Hamlet3000, 2023-02-03:
no, the green dot just jumps back to the previous setting, if I try to change something.

Hamlet3000, 2023-02-03:
And I'm not sure if it's related, but Vector crashes with error code 915 when I try set location

kercre123, 2023-02-04:
What device is wire-pod running on? May you pull logs from the robot and get them to me? (can be done by heading to [https://www.project-victor.org/noflow-devsetup](https://www.project-victor.org/noflow-devsetup), connecting the robot, and entering `logs` in the console)

Hamlet3000, 2023-02-04:
Its running on an RaspberryPi 4B with 4GB Ram.
Vectors log is attached

[vector-logs-2023-02-04-08-54-06.tar.zip](https://github.com/kercre123/wire-pod/files/10608205/vector-logs-2023-02-04-08-54-06.tar.zip)

ThommoMC, 2023-02-04:
I also Raspberry Pi 4B with 4G ram, I can confirm it works perfectly without any issues. I am able to change any of Vector’s settings without any issues.

Hamlet3000, 2023-02-04:
I tired to get back to DDL servers and restored the original server_config.json on Vector. When trying to set up Vecot via the Android app, the error 923 came up. neither a reboot nor the complete new setup via wire-pod works now. After a few seconds in which Vector shows his eyes, appears again the error 923.

Hamlet3000, 2023-02-04:
It seems that the problem was indeed with my Vector. I flattened everything again and installed the latest OSKR firmware 1.8.1. Now everything works again as it should. Please excuse the confusion

closed 2023-02-04 by Hamlet3000

iPsych, 2023-03-08:
> It seems that the problem was indeed with my Vector. I flattened everything again and installed the latest OSKR firmware 1.8.1. Now everything works again as it should. Please excuse the confusion

Hello,
I am experiencing same 923 error. How did you loaded latest OSKR firmware 1.8.1?

Hamlet3000, 2023-03-08:
I used the EscapePod from DDL to reinstall the latest OSKR firmware. There is the possibility to upload .ota files and reinstall Vector with them.
But maybe it's really just a buggy version of vic-cloud or vic-gateway.... (?)

iPsych, 2023-03-08:
> I used the EscapePod from DDL to reinstall the latest OSKR firmware. There is the possibility to upload .ota files and reinstall Vector with them. But maybe it's really just a buggy version of vic-cloud or vic-gateway.... (?)

I think so... Maybe I have to do it with latest EscapePod?
Do you have .OTA of 1.8.1? If so, can you share file/link, please?

Hamlet3000, 2023-03-08:
I think that the OSKR firmware is hard linked to the serial number of the Vector, so it is not possible to share it. My .ota would probably not work for you.

Hamlet3000, 2023-03-08:
Have you tried contacting DDL to have them resend you the latest 1.8.1 OSKR image für your robot? Steve Coblentz is always very cooperative on Facebook.

iPsych, 2023-03-08:
@Hamlet3000 Thanks. I sent a message to him. Did you ask help through facebook directly?

Hamlet3000, 2023-03-08:
@iPsych yes, I contacted Steve directly through Facebook. He usually replies within a day