kercre123/wire-pod #82: OpenAI Response is not "spoken" by Vector

ibwo88, 2023-02-09:
Hi, wondering why the OpenAI response is not processed. Thanks.
Log captured:
Feb  9 11:53:48 escapepod start.sh[2700]: Making request to OpenAI...
Feb  9 11:53:52 escapepod start.sh[2700]: OpenAI response: Check GP T stands for General Purpose Technology. It is a type of technology that can be used for a variety of purposes, such as communication, computing, entertainment, and robotics.
Feb  9 11:53:52 escapepod start.sh[2700]: (KG) Bot 1 request served.
Feb  9 11:53:52 escapepod start.sh[2700]: Knowledge graph error
Feb  9 11:53:52 escapepod start.sh[2700]: rpc error: code = Unavailable desc = transport is closing

kercre123, 2023-02-11:
My guess is that the OpenAI API request took too long and the bot closed the connection before the response made it to wire-pod. Does this happen every time?

ibwo88, 2023-02-11:
Hi Wire, first thanks so much for making this wire-pod for everyone.  Yeah, I tried many times and the error was the same.  Was tailing the log, and the messages only appeared after vector gave up. But the bot gave up really quick - seems like no waiting time.   

Feb 11 16:01:38 escapepod start.sh[483]: Bot 1 Stream type: OPUS
Feb 11 16:01:38 escapepod start.sh[483]: (Bot 1, Vosk) Processing...
Feb 11 16:01:43 escapepod start.sh[483]: (Bot 1) End of speech detected.
Feb 11 16:01:43 escapepod start.sh[483]: Bot 1 Transcribed text: question
Feb 11 16:01:43 escapepod start.sh[483]: Not a custom intent
Feb 11 16:01:43 escapepod start.sh[483]: Perfect match for intent intent_knowledge_promptquestion (question)
Feb 11 16:01:43 escapepod start.sh[483]: Checking params for candidate intent intent_knowledge_promptquestion
Feb 11 16:01:43 escapepod start.sh[483]: Bot 1 Intent Sent: intent_knowledge_promptquestion
Feb 11 16:01:43 escapepod start.sh[483]: No Parameters Sent
Feb 11 16:01:43 escapepod start.sh[483]: Bot 1 request served.
Feb 11 16:01:46 escapepod start.sh[483]: Bot 1 Stream type: OPUS
Feb 11 16:01:46 escapepod start.sh[483]: (Bot 1, Vosk) Processing...
Feb 11 16:01:51 escapepod start.sh[483]: (Bot 1) End of speech detected.
Feb 11 16:01:52 escapepod start.sh[483]: Bot 1 Transcribed text: my head who is donald trump
Feb 11 16:01:52 escapepod start.sh[483]: Making request to OpenAI...
Feb 11 16:02:07 escapepod start.sh[483]: OpenAI response: Donald Trump is the 45th President of the United States, who took office in 2017. He is a businessman and television personality who was previously the host of The Apprentice.
Feb 11 16:02:07 escapepod start.sh[483]: (KG) Bot 1 request served.
Feb 11 16:02:07 escapepod start.sh[483]: Knowledge graph error
Feb 11 16:02:07 escapepod start.sh[483]: rpc error: code = Unavailable desc = transport is closing

ibwo88, 2023-02-11:
Hey Wire, after i did a git pull and it is working now. Thanks.

closed 2023-02-11 by ibwo88