kercre123/wire-pod #83: Voice commands not working while charging.

Manic137, 2023-02-10:
Getting this weird error only while he is charging:
```
Feb 10 08:46:36 escapepod start.sh[11116]: Bot 1 Stream type: OPUS
Feb 10 08:46:36 escapepod start.sh[11116]: (Bot 1, Vosk) Processing...
Feb 10 08:46:43 escapepod start.sh[11116]: EOF
Feb 10 08:46:43 escapepod start.sh[11116]: Bot 1 Intent Sent: intent_system_noaudio
Feb 10 08:46:43 escapepod start.sh[11116]: Bot 1 Parameters Sent: map[error:EOF]
Feb 10 08:46:52 escapepod start.sh[11116]: Bot 1 Stream type: OPUS
Feb 10 08:46:52 escapepod start.sh[11116]: (Bot 1, Vosk) Processing...
Feb 10 08:46:59 escapepod start.sh[11116]: EOF
Feb 10 08:46:59 escapepod start.sh[11116]: Bot 1 Intent Sent: intent_system_noaudio
Feb 10 08:46:59 escapepod start.sh[11116]: Bot 1 Parameters Sent: map[error:EOF]
```

kercre123, 2023-02-11:
What kind of device is wire-pod running on? It seems to just be processing the text too slowly and doesn't seem to have anything to do with the charger.

Manic137, 2023-02-11:
Issue has been solved, turns out I was using old instructions. Using a new link to install the V2.0.1 firmware fixed the issue!

Manic137, 2023-02-11:
> What kind of device is wire-pod running on? It seems to just be processing the text too slowly and doesn't seem to have anything to do with the charger.

It's running on a Raspberry Pi Zero 2 W with 64bit RaspPi Lite OS

Manic137, 2023-02-11:
Scratch that, issue is back, maybe you're right about the device being too slow, but the weird thing is that issue only persists on the charger, commands and questions works perfectly the moment I take him off the charger so I don't really know what to think

kercre123, 2023-02-11:
Are the charger and the Pi connected to the same power source? Vector could be drawing current and preventing the Pi from getting its required current.

Manic137, 2023-02-11:
> Are the charger and the Pi connected to the same power source? Vector could be drawing current and preventing the Pi from getting its required current.

It is not in the same power source, separate wall plugs

Manic137, 2023-02-11:
Looks like there are times that the voice thing doesn't work outside of the charger too, I think you're right that it is a performance issue, so I'll close it here for now

closed 2023-02-11 by Manic137