kercre123/wire-pod #85: Successfully installed Cyb3rdog's SDK and authenticated with DDL servers, but unable to run scripts in wire-pod due to weird error

Manic137, 2023-02-12:
Error:
```
Feb 11 06:42:13 escapepod start.sh[8336]: (Bot 1, Vosk) Processing...
Feb 11 06:42:17 escapepod start.sh[8336]: (Bot 1) End of speech detected.
Feb 11 06:42:18 escapepod start.sh[8336]: Bot 1 Transcribed text: best
Feb 11 06:42:18 escapepod start.sh[8336]: Custom Intent Matched: test - test - intent_greeting_hello
Feb 11 06:42:18 escapepod start.sh[8336]: Executing: /bin/python3 /home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py
Feb 11 06:42:18 escapepod start.sh[8336]: exit status 1: Traceback (most recent call last):
Feb 11 06:42:18 escapepod start.sh[8336]: File "/home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py", line 22, in
Feb 11 06:42:18 escapepod start.sh[8336]: import anki_vector
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/anki_vector/init.py", line 22, in
Feb 11 06:42:18 escapepod start.sh[8336]: from . import messaging
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/init.py", line 42, in
Feb 11 06:42:18 escapepod start.sh[8336]: from . import protocol
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/protocol.py", line 24, in
Feb 11 06:42:18 escapepod start.sh[8336]: from .behavior_pb2 import *
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/behavior_pb2.py", line 16, in
Feb 11 06:42:18 escapepod start.sh[8336]: from anki_vector.messaging import messages_pb2 as anki__vector_dot_messaging_dot_messages__pb2
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/messages_pb2.py", line 17, in
Feb 11 06:42:18 escapepod start.sh[8336]: from anki_vector.messaging import response_status_pb2 as anki__vector_dot_messaging_dot_response__status__pb2
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/response_status_pb2.py", line 33, in
Feb 11 06:42:18 escapepod start.sh[8336]: _descriptor.EnumValueDescriptor(
Feb 11 06:42:18 escapepod start.sh[8336]: File "/usr/local/lib/python3.9/dist-packages/google/protobuf/descriptor.py", line 755, in new
Feb 11 06:42:18 escapepod start.sh[8336]: _message.Message._CheckCalledFromGeneratedFile()
Feb 11 06:42:18 escapepod start.sh[8336]: TypeError: Descriptors cannot not be created directly.
Feb 11 06:42:18 escapepod start.sh[8336]: If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
Feb 11 06:42:18 escapepod start.sh[8336]: If you cannot immediately regenerate your protos, some other possible workarounds are:
Feb 11 06:42:18 escapepod start.sh[8336]: 1. Downgrade the protobuf package to 3.20.x or lower.
Feb 11 06:42:18 escapepod start.sh[8336]: 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).
Feb 11 06:42:18 escapepod start.sh[8336]: More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
Feb 11 06:42:18 escapepod start.sh[8336]: Custom Intent Exec Output:
Feb 11 06:42:18 escapepod start.sh[8336]: Bot 1 Intent Sent: intent_greeting_hello
Feb 11 06:42:18 escapepod start.sh[8336]: No Parameters Sent
Feb 11 06:42:18 escapepod start.sh[8336]: This is a custom intent or plugin!
Feb 11 06:42:18 escapepod start.sh[8336]: Bot 1 request served.
```

Running the script outside of wire-pod makes it run normally:
```
 
pi@escapepod:~ $ python3 /home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py
23:58:12.952               robot.Robot WARNING  No serial number or name provided. Automatically selecting                                                                                                                                 REDACTED
23:58:12.965     connection.Connection    INFO  Connecting to REDACTED:443 for Vector-REDACTED using /home/pi                                                                                                                                 /.anki_vector/Vector-REDACTED.cert
23:58:19.725     connection.Connection    INFO  BehaviorControl control_granted_response { }
Say 'Hello World'...
```

Am I doing something wrong? Here's some images for more context: https://imgur.com/a/NnMVrvv

Manic137, 2023-02-12:
Update 1: Found a cause! Running the command as sudo causes it to fail, here's the error:
```
 sudo python3 /home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py
Traceback (most recent call last):
  File "/home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py", line 22, in <module>
    import anki_vector
  File "/usr/local/lib/python3.9/dist-packages/anki_vector/__init__.py", line 22, in <module>
    from . import messaging
  File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/__init__.py", line 42, in <module>
    from . import protocol
  File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/protocol.py", line 24, in <module>
    from .behavior_pb2 import *
  File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/behavior_pb2.py", line 16, in <module>
    from anki_vector.messaging import messages_pb2 as anki__vector_dot_messaging_dot_messages__pb2
  File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/messages_pb2.py", line 17, in <module>
    from anki_vector.messaging import response_status_pb2 as anki__vector_dot_messaging_dot_response__status__pb2
  File "/usr/local/lib/python3.9/dist-packages/anki_vector/messaging/response_status_pb2.py", line 33, in <module>
    _descriptor.EnumValueDescriptor(
  File "/usr/local/lib/python3.9/dist-packages/google/protobuf/descriptor.py", line 755, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
```

Manic137, 2023-02-12:
How would I fix this? I'm guessing it has to do with some installation issues, some being installed with sudo, some with not, sadly I can't really recall which was installed with or without sudo, and my knowledge in this in too little for me to guess what to reinstall

kercre123, 2023-02-12:
Try running the following in a shell session:

```
sudo pip uninstall protobuf
sudo pip install protobuf==3.19.6
```

Then try your command

Manic137, 2023-02-12:
> Try running the following in a shell session:
> 
> ```
> sudo pip uninstall protobuf
> sudo pip install protobuf==3.19.6
> ```
> 
> Then try your command

Oh my, didn't think you were awake, was just about to try that haha

Update 2: Found something interesting online, someone's python was a different version when they checked with and without sudo, so I tried that, but they were the same, but then it hit me, what if protobuf was different, so I checked, this was the output:
```
pi@escapepod:~ $ sudo python3 --version
Python 3.9.2
pi@escapepod:~ $ python3 --version
Python 3.9.2
pi@escapepod:~ $ pip3 show protobuf
Name: protobuf
Version: 3.19.0
Summary: Protocol Buffers
Home-page: https://developers.google.com/protocol-buffers/
Author: None
Author-email: None
License: 3-Clause BSD License
Location: /home/pi/.local/lib/python3.9/site-packages
Requires:
Required-by: googleapis-common-protos
pi@escapepod:~ $ sudo pip3 show protobuf
Name: protobuf
Version: 4.21.12
Summary: None
Home-page: https://developers.google.com/protocol-buffers/
Author: protobuf@googlegroups.com
Author-email: protobuf@googlegroups.com
License: 3-Clause BSD License
Location: /usr/local/lib/python3.9/dist-packages
Requires:
Required-by: googleapis-common-protos
```

Surely enough, it is! This is likely the cause, I shall attempt to install protobuf with sudo privledges now and get back to you

Manic137, 2023-02-12:
Everything works perfectly now! Albeit, with a slight hickup, trying to run the sdk as sudo caused it to think I am not authenticated with the DDL servers, so I had to rerun the setup for the Python SDK with sudo as well to reauthenticate with sudo, and after that I tried running the script at it worked beautifully! Thank you so much!

```
Feb 12 05:11:21 escapepod start.sh[422]: (Bot 1, Vosk) Processing...
Feb 12 05:11:25 escapepod start.sh[422]: (Bot 1) End of speech detected.
Feb 12 05:11:26 escapepod start.sh[422]: Bot 1 Transcribed text: test
Feb 12 05:11:26 escapepod start.sh[422]: Custom Intent Matched: test - test - intent_greeting_hello
Feb 12 05:11:26 escapepod start.sh[422]: Executing: /usr/bin/python3 /home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py
Feb 12 05:11:34 escapepod start.sh[422]: Custom Intent Exec Output: Say 'Hello World'...
Feb 12 05:11:34 escapepod start.sh[422]: This is a custom intent or plugin!
Feb 12 05:11:34 escapepod start.sh[422]: Bot 1 request served.
Feb 12 05:13:08 escapepod start.sh[422]: Bot 1 Stream type: OPUS
Feb 12 05:13:08 escapepod start.sh[422]: (Bot 1, Vosk) Processing...
Feb 12 05:13:12 escapepod start.sh[422]: (Bot 1) End of speech detected.
Feb 12 05:13:12 escapepod start.sh[422]: Bot 1 Transcribed text: best
Feb 12 05:13:12 escapepod start.sh[422]: Custom Intent Matched: test - test - intent_greeting_hello
Feb 12 05:13:12 escapepod start.sh[422]: Executing: /usr/bin/python3 /home/pi/vector-python-sdk/examples/tutorials/01_hello_world.py
Feb 12 05:13:19 escapepod start.sh[422]: Custom Intent Exec Output: Say 'Hello World'...
Feb 12 05:13:19 escapepod start.sh[422]: This is a custom intent or plugin!
Feb 12 05:13:19 escapepod start.sh[422]: Bot 1 request served.
```

closed 2023-02-12 by Manic137