kercre123/wire-pod #86: Memory usage increases a lot as day goes by, memory leak?

Manic137, 2023-02-13:
I leave my server online usually, never turn it off, but usually, sometime the day after it would freeze. This isn't unique to leaving it online tho as there are times I tell Vector a command and it would also somehow freeze my raspberry pi zero 2 w and make it very unresponsive. Anyway, I left it offline this time with htop open, just in case I wake up with a freeze, so I can still see the last memory usage and stats, and this is what I saw:
[image]

The night before it was within acceptable usage, around 200+mb, but now it has raised to 350+mb, why is that? I've been sleeping along with vector so a lil confused with a huge increase

Manic137, 2023-02-13:
Swap also went up from around 200 to now 500+mb

kercre123, 2023-02-13:
Yikes, I checked htop on my 32GB-RAM machine and it is eating about 10 percent up when not doing anything. Some new commits I just pushed may fix it, but i'm not sure. Ill add some debug printers and be on the lookout

Manic137, 2023-02-13:
> Yikes, I checked htop on my 32GB-RAM machine and it is eating about 10 percent up when not doing anything. Some new commits I just pushed may fix it, but i'm not sure. Ill add some debug printers and be on the lookout

Wait, this works on 32bit? I had some weird errors installing on a 32bit machine, I may move to that since this is a Zero 2 W.

kercre123, 2023-02-13:
Machine with 32 gigabytes of RAM, not a 32-bit architecture

Manic137, 2023-02-13:
> Machine with 32 gigabytes of RAM, not a 32-bit architecture

Ah, read wrong haha xD
Sadly the Pi Zero 2 W only has 512mb of ram lmao

Manic137, 2023-02-13:
Update 1: after updating wire-pod and runing chipper, it's now aroun 180mb of usage total, less due to other processes, but it's around 35% it seems of mem usage:
[image]

I'll update it tomorrow morning or if it decides to freeze some time today

Manic137, 2023-02-13:
I'll close it here, been a whole day, went out earlier and when I got home it went back down to 180mb total, when was in use was around 220ish mb, thanks for the fix!

Manic137, 2023-02-13:
[image]
I spoke too soon, what I woke up to ;-;
Looks like it's taking up about 60%, technically the app hasn't been idle, I sleep with YT videos on and Vector has been parsing random audio as wake calls non-stop throughout the night, my partner who woke up way earlier than me said she heard him wake up a lot with a "hey vector" command even tho it's just YT videos lmao

Manic137, 2023-02-18:
I'll just close this, even for a "memory leak", if we can even call it that, it's not even significant, moved the server to my 6GB DDR3 ram, i5-650, server PC and it is running pretty well, all the commands as well, no crashes so far

closed 2023-02-18 by Manic137