kercre123/wire-pod #88: setup.sh exits after VOSK install

jfancherla, 2023-02-13:
the setup.sh script dated 13 FEB exits after the VOSK setup. I am unable to complete the install due to missing files as no config file is created and setup.sh is unable to run option 6. As result setup.sh scp <vectorip> < /path/to/key> also fails due to the missing config file.

this worked in the previous version

jfancherla, 2023-02-14:
cannot onboard OSKR Vector as "sudo ./setup.sh scp vectorip /path/to/key" fails with error "server_config.json file missing. You need to generate this file with ./setup.sh's 6th option" If I then run setup.sh again it exits after setting up VOSK with message "wire-pod has been set up successfully!".

kercre123, 2023-02-15:
Pull the latest commit and follow the new wiki instructions, OSKR/dev-bot setup is now done via the web interface

jfancherla, 2023-02-16:
I did.

If you have an OSKR or dev bot, you can set them up with wire-pod via the web configuration interface. In the "Bot Setup" section, look for "Set up OSKR/dev bot", enter the bot's IP address, upload the key, then press set up

I assume that via the web configuration interface you mean the web server in wire-pod. When I select "Bot Setup”. I get instructions to pair with the robot. This never works. I have tried with Safari and with Chrome. Running MacOS, wire-pod installed on Raspberry Pi - only thing running there. Looking at the code it appears that you call some javascript for the Bot Setup. 

This javascript does not work - I get a button that appears that says “Disconnect”. Sometimes I get a button labeled with the robot name but clicking on the button does nothing. https://keriganc.com/vector-epod-setup/html/main.html works just fine for pairing. 

Should I post this to git as an issue?

kercre123, 2023-02-16:
Are you sure you are on the latest commit? I pushed some code yesterday. There should be a section under the bot authentication which is specifically for OSKR/dev-bots (should be two sections overall in Bot Setup).  Also make sure to do CTRL+SHIFT+R (or CMD on macOS) on the configuration page in order to redownload all resources.

jfancherla, 2023-02-16:
I will start all over again. I did the pull yesterday so it probably matters what time.

kercre123, 2023-03-10:
Closing due to inactivity.

closed 2023-03-10 by kercre123