kercre123/wire-pod #95: Enhancement: Trigger a new voice request for response if OpenAI responds with a question

Sovenger, 2023-02-20:
When the user has OpenAI enabled, if the parsed response from OpenAI ends in a question mark, trigger the bot to ask for new input from the user.

This would allow for a more dynamic question and response format instead of having to trigger listening again with “Hey, Vector!”.

kercre123, 2023-02-20:
I think that is something which would need to implemented robot-side. Wire-pod cannot trigger a voice request on the robot

closed 2023-02-20 by kercre123

Sovenger, 2023-02-24:
Couldn't we detect that question mark in the response from OpenAI and then assume control of Vector with the SDK and trigger a knowledge graph question, which would then cause a new connection back via Chipper?

kercre123, 2023-02-25:
I forgot that the knowledge graph requests were done via intents, and that they can be activated via the SDK interface. It is regular voice commands which can't be activated via the SDK. I played around with it a bit and it doesn't seem to work until after he stops speaking and until he is done showing his little happy face. Sending the intent request right after he is done speaking just doesn't work. We would also need to figure out exactly when to send the request, which means figuring out exactly when he is done speaking, which seems like a lot of work.