kercre123/wire-pod #97: Can't start Chipper with vosk.

iPsych, 2023-02-24:
Below error shows up when I run sudo ./chipper/start.sh
Any specific go version needed in Ubuntu 18? I followed all wiki instructions.

```
# github.com/alphacep/vosk-api/go
../../go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.45/vosk.go:6:11: fatal error: vosk_api.h: No such file or directory
 // #include <vosk_api.h>
           ^~~~~~~~~~~~
compilation terminated.
```

iPsych, 2023-02-24:
It seems that the setup.sh create .vosk folder in the /root/ directory. 
Is it normal? 
via
`sudo ls /root/.vosk/libvosk
`
I can see vosk_api.h is in there.

However, whenever I run the 
`sudo ./chipper/start.sh
`
Above error shows up.

iPsych, 2023-02-24:
I found above error is previously installed python vosk.
After pip3 uninstall vosk, the start.sh can be run, but another error happenes.

```
# command-line-arguments
/usr/local/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
/tmp/go-link-581453104/000006.o: In function `_cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_partial_words':
vosk.cgo2.c:(.text+0x4be): undefined reference to `vosk_recognizer_set_partial_words'
collect2: error: ld returned 1 exit status

```

kercre123, 2023-02-25:
What version of GCC is installed? The one installed in Ubuntu 18 might be too old for the vosk code.

iPsych, 2023-02-26:
@kercre123 gcc 9.4.0

I tested several go versions too. 
golang >12 seems work. 

```
Getting VOSK assets
vosk-linux-x86_64-0 100%[===================>]   6.85M  11.1MB/s    in 0.6s    
Archive:  vosk-linux-x86_64-0.3.43.zip
   creating: vosk-linux-x86_64-0.3.43/
  inflating: vosk-linux-x86_64-0.3.43/vosk_api.h  
  inflating: vosk-linux-x86_64-0.3.43/libvosk.so  
go: downloading github.com/alphacep/vosk-api v0.3.45
go: downloading github.com/alphacep/vosk-api/go v0.3.45

wire-pod has been set up successfully!
```

Than I tested both gcc-9 and gcc-11 but no luck.
same error as below.
```
../../go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.45/vosk.go:6:11: fatal error: vosk_api.h: No such file or directory
    6 | // #include <vosk_api.h>
      |           ^~~~~~~~~~~~
compilation terminated.

```
Would you let me know the 'tested' setting, especially Raspberry pi 4?
(i.e. Ubuntu 20, gcc>10, Raspberry Pi OS > X.XX with gcc X.XX, go X.XX.)

iPsych, 2023-02-26:
Got it.
Above error was due to mistakenly copied libvosk.so in /usr/local/lib.
The proper dir is /usr/lib/

closed 2023-02-26 by iPsych