kercre123/wire-pod #98: Can't activate vector

a7ypically, 2023-02-24:
I'm using an out of the box 2.0 vector. Never activated with ddl.
wire-pod is running on rpi 4.
I followed the instructions but my vector can't seem to connect to wire-pod.
Using a sniffer I can see no connection attempts at all from vector to wire-pod. It does connect to the Internet and pulls data.
It looks like an issue with mDNS. I pulled the logs from vector and see these:
vic-cloud: Token queue error: rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp: lookup escapepod.local: no such host"
avahi is running on the rpi4. I can resolve and ping escapepod.local fine from a client on the same wifi as Vector. I can also see using a sniffer that Vector sends mDNS requests for escapepod.local and I can see a reply from the rpi with an answer.
Vector's firmware is v2.0.1.6076

Any ideas?

Thanks.

kercre123, 2023-03-10:
Is wire-pod fully set up? It will only accept connections if it shows `wire-pod started successfully!` in its logs. May you send over all logs from wire-pod and from the bot?

kercre123, 2023-04-16:
Closing due to inactivity.

closed 2023-04-16 by kercre123