kercre123/wire-pod #102: OpenAi returned no response

Hotline2020, 2023-03-06:
I freshly built Wire-Pod yesterday. So far everything works except for one thing. As soon as I make my request via "openai" I get "OpenAI returned no response." return. I've reset my OpenAI API several times and even switched my free account to a paid account. Does anyone else have problems connecting to Openai. It works fine via Houndify.

kercre123, 2023-03-18:
The only cause I can think of for that is a bad API key.

ophidiasnaketongue, 2023-04-02:
I've just started getting this same error. I've tried restarting the wire-pod and the router, and generated a new key. I am still getting the error- and even more strangely, no attempt to access the knowledge graph is showing in the wire pod logs.

Sovenger, 2023-04-02:
Having the same issue.

ophidiasnaketongue, 2023-04-02:
SOLVED- I've figured out what had happened. My free trial OpenAI account and its attached credits had expired, so I had to set up a paid account and generate a new API key. Thankfully, the accounts are very cheap.

Sovenger, 2023-04-04:
Yep, same for me. My trial tokens expired. I went to a paid plan and now the issue is resolved.

@kercre123 is there any error code or message you get back from the server that indicates the trial has ended so that we can have Vector speak "Open AI has indicated your free trial period has ended. Please move to an Open AI pay-as-you-go plan if you want to use VOSK + Knowledge Graph + Open AI. Otherwise, switch back to Houndify"?

khanhnguyen12345, 2023-06-03:
I have a question. why i say "hey vector" but it doesn't work. I have to press the button on its back

Stoj26, 2023-08-06:
Same issue - have a paid OpenAI account and everything.

kercre123, 2023-11-06:
Should be fine now

closed 2023-11-06 by kercre123