kercre123/wire-pod #106: Sodium errors in Raspberry Pi 4 running latest Raspbian

nicholas-johnson, 2023-03-09:
I'm getting errors when running sudo ./chipper/start.sh. Looks like an issue with the Sodium Go wrapper

```
/root/go/pkg/mod/github.com/jamesruan/sodium@v0.0.0-20181216154042-9620b83ffeae/generichash.go:56:72: cannot use _Ctype_ulong(cryptoGenericHashStateBytes) (value of type _Ctype_ulong) as type _Ctype_uint in argument to (_Cfunc_sodium_malloc)
/root/go/pkg/mod/github.com/jamesruan/sodium@v0.0.0-20181216154042-9620b83ffeae/generichash.go:71:72: cannot use _Ctype_ulong(cryptoGenericHashStateBytes) (value of type _Ctype_ulong) as type _Ctype_uint in argument to (_Cfunc_sodium_malloc)
/root/go/pkg/mod/github.com/jamesruan/sodium@v0.0.0-20181216154042-9620b83ffeae/generichash.go:99:74: cannot use _Ctype_ulong(cryptoGenericHashStateBytes) (value of type _Ctype_ulong) as type _Ctype_uint in argument to (_Cfunc_sodium_malloc)
/root/go/pkg/mod/github.com/jamesruan/sodium@v0.0.0-20181216154042-9620b83ffeae/sign.go:226:67: cannot use _Ctype_ulong(cryptoSignStateBytes) (value of type _Ctype_ulong) as type _Ctype_uint in argument to (_Cfunc_sodium_malloc)
```

Device is a genuine Raspberry Pi 4 with 8Gb (bought from the official Raspberry Pi store) running Raspbian Bullseye.

```pi@escapepod:~/wire-pod $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I'm a software engineer, but have never used Go, so not entirely sure what I'm looking at here. Looks like a typing issue. I note this line in go.mod, which I'm guessing is the offending module: 	

    github.com/jamesruan/sodium v0.0.0-20181216154042-9620b83ffeae // indirect

How to proceed please?

nicholas-johnson, 2023-03-09:
Ran go get github.com/jamesruan/sodium in chipper which has upgraded it to 1.0.14 and it seems to be running now. Will test. Would you like a PR?

nicholas-johnson, 2023-03-09:
Seems to be working alright now with the upgraded sodium module.

kercre123, 2023-03-10:
Is the OS 32-bit or 64-bit? A PR would be appreciated.

nicholas-johnson, 2023-03-12:
PR is here: https://github.com/kercre123/wire-pod/pull/111

Seems to work on my machine. A fresh install worked and is driving my vector.

```
pi@escapepod:~/wire-pod $ uname -m
armv7l
```

So I think that's 32 bit.

closed 2023-04-16 by kercre123