kercre123/wire-pod #108: Issue getting Go SDK wrapper

BillMerryman, 2023-03-10:
Trying to get the SDK wrapper:

sudo /usr/local/go/bin/go get github.com/kercre123/vector-go-sdk/pkg/sdk-wrapper

go: github.com/kercre123/vector-go-sdk@v0.0.0-20221204234153-150e0127b2c9: parsing go.mod:
	module declares its path as: github.com/fforchino/vector-go-sdk
	        but was required as: github.com/kercre123/vector-go-sdk

was able to get it to build with the fforchino package, but when trying to run it get:

2023.03.10 14:49:52: Error loading plugin: plugin.so
2023.03.10 14:49:52: plugin.Open("./plugins/plugin"): plugin was built with a different version of package github.com/fforchino/vector-go-sdk/pkg/vectorpb

kercre123, 2023-03-10:
What happens when you try using github.com/fforchino/vector-go-sdk/pkg/sdk-wrapper instead? fforchino's is much more up-to-date than my version.

BillMerryman, 2023-03-10:
That's the:

2023.03.10 14:49:52: Error loading plugin: plugin.so
2023.03.10 14:49:52: plugin.Open("./plugins/plugin"): plugin was built with a different version of package github.com/fforchino/vector-go-sdk/pkg/vectorpb

error I was getting. It would build, but I got this message when trying to run it. The version of Wirepod I have installed is the wire-prod-pod. I built it first. Do I have to mess with the dependencies in wire-prod-pod and rebuild it?

kercre123, 2023-03-10:
wire-prod-pod will be deprecated soon. I believe I have set all paths to fforchino's in the current version of wire-pod, but some remnants of mine may remain in wire-prod-pod. I would remove wire-prod-pod and try installing wire-pod (see the wiki on this repo).

closed 2023-04-16 by kercre123