kercre123/wire-pod #110: Just upgraded wire-pod- cannot authenticate robot

ophidiasnaketongue, 2023-03-11:
Hi there,
I have just upgraded wire-pod and successfully completed the set up process. My Vector can use Chipper, but it's not able to run custom intents and I can't set his location because he won't connect fully with the setup process. I am running wire-pod on a raspberry pi model 4b.

When I try to connect Vector using Bluetooth via the web server, I place Vector into pairing mode, and then tap 'begin scanning' in the Wire-Pod interface. When I do this, the name of my Vector comes up. When I click on it, it says 'connecting to robot', but nothing further happens. Vector does not generate a bluetooth key and remains with the ##### message on his screen before eventually timing out.

It's a very minor thing, but I'd just like him to use Centigrade instead of farenheit :D.

ophidiasnaketongue, 2023-03-14:
I've got a bit further with this- I found that not using Chrome by accident borked everything, but when I restarted both the raspberry pi and Vector, I could then connect him to wire-pod.

However, I can't find the setup area to get him to use centigrade. It used to be there but it's missing now? Anyone know how I can get hold of that?

kercre123, 2023-03-18:
In the web interface, there should be an option for "Bot Settings" which should let you change stuff like that.

ophidiasnaketongue, 2023-03-18:
Thanks for your reply. I was getting an error message when I connected to that page, which I presumed was because my robot isn't an SDK one. However, it is mysteriously working now. No idea what has changed!

closed 2023-04-16 by kercre123