kercre123/wire-pod #116: Can't access SDK from plugin

BillMerryman, 2023-03-15:
I apologize if this is the wrong place to post.  I got a plugin working, now trying to access the SDK from plugin. I pared it back down to a real simple example that produces errors. The InitSDK appears to be succeeding, but produces an 'INI file missing' message in the log. The sdk_wrapper.Robot.BehaviorControl call then errors out.

Here is my code:

package main

import (
	sdk_wrapper "github.com/fforchino/vector-go-sdk/pkg/sdk-wrapper"
	"context"
)

var Utterances = []string{"test plugin"}
var Name = "Test Plugin"

func Action(transcribedText string, botSerial string) string {

	sdk_wrapper.InitSDK(botSerial)
	ctx := context.Background()
	start := make(chan bool)
	stop := make(chan bool)

	go func() {
		_ = sdk_wrapper.Robot.BehaviorControl(ctx, start, stop)
	}()

	for {
		select {
		case <-start:
			stop <- true
			break
		}
	}

	return "intent_imperative_praise"

}

and my logged errors:

Mar 14 21:13:18 escapepod start.sh[2105]: Bot 1 Transcribed text: test plugin
Mar 14 21:13:18 escapepod start.sh[2105]: Text matched plugin Test Plugin, executing function
Mar 14 21:13:18 escapepod start.sh[2105]: 2023/03/14 21:13:18 INI file missing
Mar 14 21:13:18 escapepod start.sh[2105]: panic: runtime error: invalid memory address or nil pointer dereference
Mar 14 21:13:18 escapepod start.sh[2105]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x7f04e81298]
Mar 14 21:13:18 escapepod start.sh[2105]: goroutine 11 [running]:
Mar 14 21:13:18 escapepod start.sh[2105]: github.com/fforchino/vector-go-sdk/pkg/vector.(*Vector).BehaviorControl(0x0?, {0x1419a40?, 0x40000d6040?}, 0x0?, 0x0?)
Mar 14 21:13:18 escapepod start.sh[2105]:         /root/go/pkg/mod/github.com/fforchino/vector-go-sdk@v0.0.0-20221216080054-dfe1a258d335/pkg/vector/behaviorcontrol.go:13 +0x28
Mar 14 21:13:18 escapepod start.sh[2105]: plugin/unnamed-76c27f1c903e09f75de4f664b05a7b510424ddac.Action.func1()
Mar 14 21:13:18 escapepod start.sh[2105]:         /home/pi/wire-pod/chipper/plugins-dev/plugin.go:20 +0x38
Mar 14 21:13:18 escapepod start.sh[2105]: created by plugin/unnamed-76c27f1c903e09f75de4f664b05a7b510424ddac.Action
Mar 14 21:13:18 escapepod start.sh[2105]:         /home/pi/wire-pod/chipper/plugins-dev/plugin.go:19 +0xdc

scamfield, 2023-03-15:
It's possible that the issue you're encountering is related to the missing INI file. As mentioned in the [wire-pod wiki](https://github.com/kercre123/wire-pod/wiki), when a bot is authenticated, it's added or modified in the sdk_config.ini file, which allows the Python SDK to continue working.

In the [Vector Go SDK repository](https://github.com/kercre123/vector-go-sdk), it's mentioned that you need to create the INI file in the home directory under a .anki_vector folder. This could be the cause of the issue you're experiencing?

BillMerryman, 2023-03-16:
Yes! Thank you, that was it. 

For those who may follow behind me, I found my .anki_vector folder on my PC (from when I configured the Python SDK there), copied it to my chipper folder in the wirepod folder on my Pi, and updated the certificate path in the sdk_config.ini. 

I apologize for my RTFM moment.

closed 2023-03-16 by BillMerryman