kercre123/wire-pod #120: Error initializing bluetooth on the device running wire-pod. Use the following site on any machine with Bluetooth for setup

F4310, 2023-03-31:
[image]
I trie on two computers and  with a android smartphone. Any idea? 
My hostname alredy is scapepod.

f0rky, 2023-04-05:
I'm in the same boat, confirmed the host of wire-pod is escapepod
avahi is running on the host and is named escapepod.local
ping escapepod.local resolves to the correct IP
navigating to http://escapepod.local brings up the web menu

The instructions are pretty lite in terms of troubleshooting at this step, does anyone know what else could be checked?

F4310, 2023-04-07:
I solved the issue, try: 

Before start i did a reset on it
1 - I installed Xubuntu and followed installation guide
2 - Host name changed to escapepod
3 - Vector and server connected on the same network

closed 2023-04-07 by F4310