kercre123/wire-pod #122: What model does the openai mode use?

Jetpackjules, 2023-04-07:
GPT 3-5 or GPT3?

I am asking because 3-5 can produce much better conversational results, and costs 10x less than GPT 3

Sovenger, 2023-04-07:
https://github.com/kercre123/wire-pod/blob/7117ad9042ab2545b1eeef34e314a69335a7a1fa/chipper/pkg/wirepod/preqs/knowledgegraph.go#L77

Jetpackjules, 2023-04-07:
Thanks! While I have you, what would you reccomend as a cheaper alternative to the rasberry pi 4 if there is one?

kercre123, 2023-04-16:
wire-pod works well on an Orange Pi 4 LTS.

bliteknight, 2023-04-22:
> Thanks! While I have you, what would you reccomend as a cheaper alternative to the rasberry pi 4 if there is one?

I sell a cheaper alternative, using an Android TV box (T95mini)
https://techshop82.com/shop/wire-pod-server/

closed 2023-05-24 by kercre123