kercre123/wire-pod #124: Wire Pod on Web Server - Communication

F4310, 2023-04-07:
Hello.

I installed wire pod in a VPS (Private Virtual Server), i'm trying put Vector to communication through Word Wide Web.
Somebody know how change the communication adress that Vector use to server ?

kercre123, 2023-04-16:
Are you using a production (regular) bot or an OSKR/dev-unlocked bot? The hostname `escapepod.local` is hard-coded into production bots and it cannot be changed, so you must find a way to forward that to your VPS.

F4310, 2023-04-16:
Hello, my noble, thank for response.
It is regular, but i think can turn it on OSKT with this https://randym32.github.io/Anki.Vector.Documentation/how-to/OSKR%20Tutorial%20Windows%2010.html

closed 2023-04-16 by F4310