kercre123/wire-pod #131: Chipper

mdinolfo123, 2023-04-23:
I updated today, and ran "sudo ./chipper/start.sh"

This happened:

# github.com/kercre123/chipper/pkg/wirepod/config-ws
pkg/wirepod/config-ws/webserver.go:328:2: syntax error: unexpected else, expecting }

It won't run.

SummerSigh, 2023-04-23:
I fixed the issue. In webserver.go the DownloadVoskModel function should look like the following:

```
func DownloadVoskModel(language string) {
	filename := "vosk-model-small-"
	if language == "en-US" {
		filename = filename + "en-us-0.15.zip"
	} else if language == "it-IT" {
		filename = filename + "it-0.22.zip"
	} else if language == "es-ES" {
		filename = filename + "es-0.42.zip"
	} else if language == "fr-FR" {
		filename = filename + "fr-0.22.zip"
	} else if language == "de-DE" {
		filename = filename + "de-0.15.zip"
	} else if language == "pt-BR" {
		filename = filename + "pt-0.3.zip"
	} else {
		logger.Println("Language not valid? " + language)
		return
	}
	os.MkdirAll("../vosk", 0755)
	url := "https://alphacephei.com/vosk/models/" + filename
	filepath := os.TempDir() + "/" + filename
	destpath := "../vosk/models/" + language + "/"
	DownloadFile(url, filepath)
	UnzipFile(filepath, destpath)
	os.Rename(destpath+strings.TrimSuffix(filename, ".zip"), destpath+"model")
	vars.DownloadedVoskModels = append(vars.DownloadedVoskModels, language)
	DownloadStatus = "Reloading voice processor"
	vars.APIConfig.STT.Language = language
	vars.APIConfig.PastInitialSetup = true
	vars.WriteConfigToDisk()
	processreqs.ReloadVosk()
	logger.Println("Reloaded voice processor successfully")
	DownloadStatus = "success"
}
```

SummerSigh, 2023-04-23:
Was simply the else statement for if the "Language not valid?" not being formatted properly

mdinolfo123, 2023-04-23:
Unfortunately, I can only open webserver.go as "read only".

kercre123, 2023-04-23:
May someone open up a pull request? I won't be at a computer for a while.

mdinolfo123, 2023-04-23:
I never did a pull request. Wish I could help. Instead, I'll wait.

SummerSigh, 2023-04-23:
I got it

SummerSigh, 2023-04-23:
@mdinolfo123 the repo has just been updated. If the error goes away could you close the issue?

mdinolfo123, 2023-04-23:
@SummerSigh When it gets to "wire-pod started successfully!" it's frozen, I think. No voice commands. I can access localhost:8080 and play with that just fine. 

Do you have voice commands?

SummerSigh, 2023-04-23:
I haven't gotten bluetooth working 100% just yet

SummerSigh, 2023-04-23:
> @SummerSigh When it gets to "wire-pod started successfully!" it's frozen, I think. No voice commands. I can access localhost:8080 and play with that just fine.
> 
> Do you have voice commands?

Could you close this and make a new issue then?

closed 2023-04-23 by mdinolfo123