kercre123/wire-pod #133: Chipper Issue

mdinolfo123, 2023-04-23:
When running:

sudo ./chipper/start.sh

Everything loads according until the end:

Opening VOSK model (../vosk/models/en-US/model)
VOSK initiated successfully
Loading plugins
Starting webserver at port 8080 (http://localhost:8080)
Starting SDK app
Starting server at port 80 for connCheck
Configuration page: http://192.168.1.88:8080
Initiating TLS listener, cmux, gRPC handler, and REST handler
Starting chipper server at port 443
Starting chipper server at port 8084 for 2.0.1 compatibility
wire-pod started successfully!

and then nothing. No GUID or whatever verification stuff after "wire-pod started successfully!" Normally, there are a few more lines that read after that. In turn, there are no voice commands. localhost:8080 wire-pod configuration site works though.

SummerSigh, 2023-04-23:
What setup are you on?

mdinolfo123, 2023-04-24:
I appreciate it, but I think I'm going to have to wait for Wire. I updated merge request #130 "secondtruth patch intents". Chipper wouldn't run. You corrected the syntax error merge request #132, and still chipper won't run. 

It was fine before. I'm in no hurry.

kercre123, 2023-04-24:
It seems like it started correctly... you will only see verification stuff when a bot makes a token request, which has likely happened already in previous runtimes. It just seems that the bot is not able to reach the server, which could be due to a variety of reasons. Is your bot a production bot? If so, are you able to reach the configuration page at "escapepod.local:8080"?

mdinolfo123, 2023-04-24:
Yes, it's a production bot. Yes, I'm able to reach the configuration page. Even python scripts through SDK in terminal work.

mdinolfo123, 2023-04-24:
It's missing a few lines after "wire-pod started successfully!". Normally there are lines that read stuff about GUID or tokens or something. This is exactly how it looks after running chipper/start.sh

Initializing variables
SDK info path: /home/darkhorse/.anki_vector/
API config successfully read
Loaded jdocs file
Loaded bot info file, known bots: [008068f4]
Loaded custom intents:
Cube Pounce
Be Careful
Thank You
Pound
Stop it
Stay
Praise Ext. 
Adventure Time
test
Sweet Dreams
Go Play
Do it!
Loaded 54 intents and 54 matches (language: en-US)
Initiating vosk voice processor with language en-US
Opening VOSK model (../vosk/models/en-US/model)
VOSK initiated successfully
Loading plugins
Starting SDK app
Starting server at port 80 for connCheck
Initiating TLS listener, cmux, gRPC handler, and REST handler
Configuration page: http://192.168.1.88:8080
Starting webserver at port 8080 (http://localhost:8080)
Starting chipper server at port 443
Starting chipper server at port 8084 for 2.0.1 compatibility
wire-pod started successfully!

mdinolfo123, 2023-04-24:
Ah, the jdocs stuff loaded up. 

My production Vector is working now. Thank you, and sorry for the trouble. I'm going to close this out.

closed 2023-04-24 by mdinolfo123