kercre123/wire-pod #134: Issues connecting to vector

SummerSigh, 2023-04-24:
Currently getting error with Vector V1:

```
Error logging in. The bot is likely unable to communicate with your wire-pod instance. Make sure you followed all of the steps and try again.
```
Im hosting my wire-pod instance on an OrangePi4 with the debian buster image.

SummerSigh, 2023-04-24:
Here is the full start up for wire-pod:

```
orangepi@escapepod:~/wire-pod$ sudo chipper/start.sh 
Initializing variables
SDK info path: /home/orangepi/.anki_vector/
API config successfully read
open ../vosk/models/: no such file or directory
Loaded 54 intents and 54 matches (language: en-US)
Initiating vosk voice processor with language en-US
Opening VOSK model (../vosk/models/en-US/model)
ERROR (VoskAPI:Model():model.cc:122) Folder '../vosk/models/en-US/model' does not contain model files. Make sure you specified the model path properly in Model constructor. If you are not sure about relative path, use absolute path specification.
VOSK initiated successfully
Loading plugins
Starting webserver at port 8080 (http://localhost:8080)
Starting SDK app
Starting server at port 80 for connCheck
Initiating TLS listener, cmux, gRPC handler, and REST handler
Configuration page: http://172.20.10.3:8080
Starting chipper server at port 443
Starting chipper server at port 8084 for 2.0.1 compatibility
wire-pod started successfully!
```

mdinolfo123, 2023-04-24:
That is exactly what happened with me after #132. 

I updated, and did a hard reset on Vector and the Home Server. After running sudo chipper/start.sh, it took a while (several minutes), but eventually the jdocs and tokens loaded up right after "wire-pod started successfully!". I hope it finally loaded up for you as well.

SummerSigh, 2023-04-24:
I didn't wait around that long. Ill try again today.

F4310, 2023-04-26:
Vector tries to connect on escapepod.local, not by IP Address.
Maybe it dont reach the host, try :

- In your router, look for Static DNS 
- Set the host name (escapepod.local) and  IP address (computer running Wire-pod)
- Apply and reboot the router.

SummerSigh, 2023-04-26:
Ok! Thanks I’ll try that out!

bliteknight, 2023-05-02:
Check to see if you have avahi-daemon installed and running. That is what helps broadcast on your network what "escapepod" and "escapepod.local" is, so vector can find it. A simple test on the computer you are installing the firmware from is to check if this address is reachable

http://escapepod.local/setup.html

SummerSigh, 2023-05-07:
Hah! I solved it! It seems an older setup i was working on had the same name "escapepod"! after changing the name it worked!!! Very happy. thank you all so much!

closed 2023-05-07 by SummerSigh

SummerSigh, 2023-05-07:
Also let me know anyway I can help! Im an AI engineer working on OpenAssistant with LAION-AI! Would be more than happy to help get something like that working.