kercre123/wire-pod #135: Modify .go file

WALKINGFLOW, 2023-04-26:
I have modified openaiRequest function in knowledgegraph.go (in wire-pod\chipper\pkg\wirepod\preqs) to use gpt-3.5 model.

But the answer from my vector seems not changed.

Do I have to compile the .go file? (I'm new in go language)

What should I do for applying the change of .go file in working wire-pod? thanks in advance!

secondtruth, 2023-04-26:
Yes, you need to recompile chipper. I created a little script to do this. It works for me.

1. In SSH change to wire-pod directory
2. Save this as `build.sh` in the wire-pod directory:
```
#!/bin/bash
set -e

echo

if [[ $EUID -ne 0 ]]; then
    echo "This script must be run as root. Try 'sudo ./build.sh'."
    exit 1
fi

if [[ ! -d ./chipper ]]; then
    echo "Script is not running in the wire-pod/ directory or chipper folder is missing. Exiting."
    exit 1
fi

if [[ ! -f ./chipper/source.sh ]]; then
    echo "You need to make a source.sh file. This can be done with the setup.sh script, option 6."
    exit 1
fi
source ./chipper/source.sh

cd chipper

if [[ $STT_SERVICE == "leopard" ]]; then
    echo "building chipper with Picovoice STT service..."
    /usr/local/go/bin/go build cmd/leopard/main.go
elif [[ $STT_SERVICE == "vosk" ]]; then
    echo "building chipper with VOSK STT service..."
    export CGO_ENABLED=1
    export CGO_CFLAGS="-I$HOME/.vosk/libvosk"
    export CGO_LDFLAGS="-L $HOME/.vosk/libvosk -lvosk -ldl -lpthread"
    export LD_LIBRARY_PATH="$HOME/.vosk/libvosk:$LD_LIBRARY_PATH"
    /usr/local/go/bin/go build cmd/vosk/main.go
else
    echo "building chipper with Coqui STT service..."
    export CGO_LDFLAGS="-L$HOME/.coqui/"
    export CGO_CXXFLAGS="-I$HOME/.coqui/"
    export LD_LIBRARY_PATH="$HOME/.coqui/:$LD_LIBRARY_PATH"
    /usr/local/go/bin/go build cmd/coqui/main.go
fi

mv main chipper

echo "./chipper/chipper has been built!"
```
3. Run `chmod +x build.sh` and `./build.sh`

WALKINGFLOW, 2023-04-26:
Wow, Thank you a million!

So, recomile chipper file, and restart wire-pod service then it works, right?

It works as you said, thank you!

closed 2023-04-26 by WALKINGFLOW

secondtruth, 2023-04-26:
Yes, exactly.