kercre123/wire-pod #137: You must set PICOVOICE_APIKEY to a value.

jackseverity, 2023-05-01:
Hi there, i understand wire pod isnt made for pi3's but i was wanting to try to get it working again on my 3 so i can free my pi4 up again. 

my issue is after selecting Picovoice it asks me to 'Enter your access key' which i do and its says 'wire pod has been set up successfully'
After that i run 'sudo ./chipper/start.sh' and i get this
'Initializing variables
SDK info path: /home/pi/.anki_vector/
API config successfully read
Loaded jdocs file
Loaded bot info file, known bots: [0gf6jj64e]
Loaded 54 intents and 54 matches (language: en-US)
Initiating leopard voice processor with language en-US
You must set PICOVOICE_APIKEY to a value.

I dont understand what to do here.
Many thanks in advance :)

bliteknight, 2023-05-02:
Check that this file exist

/chipper/pico.key

And that it contains

export PICOVOICE_APIKEY=<your picoKey>

If it is not there, you could have write permission issues in your setup for what you installed wire-pod and might have to fix that.

jackseverity, 2023-05-02:
Thank you for your response BliteKnight. i checked that and yes i have that file and it contains what you wrote with my api key as well. maybe something happened. Ill start the process again and see if i have better luck. Thanks for your time

kercre123, 2023-05-07:
Just fixed this in the latest commit, pull it and try again

jackseverity, 2023-05-12:
Many thanks. works perfect now :)

closed 2023-05-12 by jackseverity