kercre123/wire-pod #138: sdk_config.ini only created with clearing user data

bliteknight, 2023-05-03:
I have been playing with the plugins implementation (going the go  route as I want to learn the language) and I noticed something when trying to use the vector-go-sdk. The sdk_conf.ini does not exist on my machine in the <user home>/.anki_vector folder.

So I thought I would use the InitSDKForWirepod but then looking through the code it was failing to get the settings for my Vector, so I looked and saw it was failing with an rpc error somewhere that's when I decided to try wiping the user data off one of my Vectors and trying again. 

That when I noticed this behavior

If the user data has been wiped and you have a new bluetooth ID, I get this output from the logs

```
May 03 11:24:28 escapepod start.sh[21476]: Closing SDK connection for 00a1151d, source: connTimer
May 03 11:25:29 escapepod start.sh[21476]: No connCheck from 192.168.0.70:443 in more than 10 seconds, will ping jdocs on next check
May 03 11:26:27 escapepod start.sh[21476]: Connecting to 00a1151d with GUID PR7kY0F8HjohEeQ4+V1ToA==
May 03 11:26:30 escapepod start.sh[21476]: Connecting to 00a1151d with GUID PR7kY0F8HjohEeQ4+V1ToA==
May 03 11:27:10 escapepod start.sh[21476]: Starting jdocs pinger timer for 192.168.0.70:443
May 03 11:27:12 escapepod start.sh[21476]: Error pinging jdocs, likely unauthenticated
May 03 11:27:21 escapepod start.sh[21476]: Token: Incoming Associate Primary User request
May 03 11:27:21 escapepod start.sh[21476]: Current time: 2023-05-03T11:27:21.632010451-05:00
May 03 11:27:21 escapepod start.sh[21476]: Token expires: 2023-05-04T11:27:21.632019452-05:00
May 03 11:27:21 escapepod start.sh[21476]: ESN not found in store or this is an associate primary user request, act as if this is a new robot
May 03 11:27:21 escapepod start.sh[21476]: Adding 192.168.0.70 to TokenHashStore
May 03 11:27:21 escapepod start.sh[21476]: UUID for this token request: 8af60060-8ee2-4e11-8e47-65bd85387323
May 03 11:27:22 escapepod start.sh[21476]: Jdocs: Incoming ReadDocs request, Robot ID: vic:00a1151d, Item(s) to return:
May 03 11:27:22 escapepod start.sh[21476]: [doc_name:"vic.AppTokens"]
May 03 11:27:22 escapepod start.sh[21476]: App tokens jdoc not found for this bot, trying bots in TokenHashStore
May 03 11:27:22 escapepod start.sh[21476]: GUID and hash successfully written for 00a1151d
May 03 11:27:22 escapepod start.sh[21476]: ReadJdocs: bot 00a1151d matched with IP 192.168.0.70 in token store
May 03 11:27:22 escapepod start.sh[21476]: Removing 192.168.0.70 from temporary token-hash store
May 03 11:27:22 escapepod start.sh[21476]: Outputting session cert to /root/.anki_vector/Vector-U8G4-00a1151d.cert
May 03 11:27:22 escapepod start.sh[21476]: Creating /root/.anki_vector/ directory
May 03 11:27:22 escapepod start.sh[21476]: WriteToIniPrimary: ESN did not match any section in sdk config file, creating
May 03 11:27:22 escapepod start.sh[21476]: WriteToIniPrimary: successfully wrote INI
May 03 11:27:22 escapepod start.sh[21476]: Removing 192.168.0.70:47306 from cert-write store
May 03 11:27:22 escapepod start.sh[21476]: Session certificate successfully output
May 03 11:27:22 escapepod start.sh[21476]: Jdocs: Incoming ReadDocs request, Robot ID: vic:00a1151d, Item(s) to return:
May 03 11:27:22 escapepod start.sh[21476]: [doc_name:"vic.RobotSettings" doc_name:"vic.RobotLifetimeStats" doc_name:"vic.AccountSettings" doc_name:"vic.UserEntitlements"]
May 03 11:27:22 escapepod start.sh[21476]: Jdocs: Incoming WriteDoc request, Item to write: vic.RobotSettings, Robot ID: vic:00a1151d
May 03 11:27:23 escapepod start.sh[21476]: Jdocs: Incoming WriteDoc request, Item to write: vic.RobotLifetimeStats, Robot ID: vic:00a1151d
May 03 11:27:23 escapepod start.sh[21476]: Jdocs: Incoming WriteDoc request, Item to write: vic.AccountSettings, Robot ID: vic:00a1151d
May 03 11:27:23 escapepod start.sh[21476]: Jdocs: Incoming WriteDoc request, Item to write: vic.UserEntitlements, Robot ID: vic:00a1151d
May 03 11:27:38 escapepod start.sh[21476]: Jdocs: Incoming WriteDoc request, Item to write: vic.AccountSettings, Robot ID: vic:00a1151d
May 03 11:27:40 escapepod start.sh[21476]: Jdocs: Incoming WriteDoc request, Item to write: vic.RobotSettings, Robot ID: vic:00a1151d
```
But if the firmware has been flashed and I am just activating a new bot on the server, I get this output
```
May 03 13:45:44 escapepod start.sh[2707]: Jdocs: Incoming ReadDocs request, Robot ID: vic:003098b1, Item(s) to return:
May 03 13:45:44 escapepod start.sh[2707]: [doc_name:"vic.RobotSettings" doc_name:"vic.RobotLifetimeStats" doc_name:"vic.AccountSettings" doc_name:"vic.UserEntitlements"]
May 03 13:45:44 escapepod start.sh[2707]: Jdocs: Incoming WriteDoc request, Item to write: vic.RobotSettings, Robot ID: vic:003098b1
May 03 13:45:44 escapepod start.sh[2707]: Jdocs: Incoming WriteDoc request, Item to write: vic.RobotLifetimeStats, Robot ID: vic:003098b1
May 03 13:45:45 escapepod start.sh[2707]: Jdocs: Incoming WriteDoc request, Item to write: vic.AccountSettings, Robot ID: vic:003098b1
May 03 13:45:46 escapepod start.sh[2707]: Jdocs: Incoming WriteDoc request, Item to write: vic.UserEntitlements, Robot ID: vic:003098b1
May 03 13:46:44 escapepod start.sh[2707]: Token: Incoming Associate Secondary Client request
May 03 13:46:44 escapepod start.sh[2707]: Current time: 2023-05-03T13:46:44.254805907-05:00
May 03 13:46:44 escapepod start.sh[2707]: Token expires: 2023-05-04T13:46:44.254826282-05:00
May 03 13:46:44 escapepod start.sh[2707]: ESN not found in store or this is an associate primary user request, act as if this is a new robot
May 03 13:46:44 escapepod start.sh[2707]: Adding 192.168.0.131 to TokenHashStore
May 03 13:46:44 escapepod start.sh[2707]: UUID for this token request: 4b3425d0-ecf9-4039-af88-e70f3ae402e3
May 03 13:46:44 escapepod start.sh[2707]: Jdocs: Incoming ReadDocs request, Robot ID: vic:003098b1, Item(s) to return:
May 03 13:46:44 escapepod start.sh[2707]: [doc_name:"vic.AppTokens"]
May 03 13:46:44 escapepod start.sh[2707]: Adding 003098b1 to bot info store
May 03 13:46:44 escapepod start.sh[2707]: App tokens jdoc not found for this bot, trying bots in TokenHashStore
May 03 13:46:44 escapepod start.sh[2707]: GUID and hash successfully written for 003098b1
May 03 13:46:44 escapepod start.sh[2707]: ReadJdocs: bot 003098b1 matched with IP 192.168.0.131 in token store
May 03 13:46:44 escapepod start.sh[2707]: Removing 192.168.0.131 from temporary token-hash store
May 03 13:46:48 escapepod start.sh[2707]: Starting jdocs pinger timer for 192.168.0.131:443
May 03 13:46:49 escapepod start.sh[2707]: Successfully got jdocs from 003098b1
May 03 13:46:57 escapepod start.sh[2707]: Jdocs: Incoming WriteDoc request, Item to write: vic.AccountSettings, Robot ID: vic:003098b1
May 03 13:46:59 escapepod start.sh[2707]: Jdocs: Incoming WriteDoc request, Item to write: vic.RobotSettings, Robot ID: vic:003098b1
```

In the second case the **WriteToIniPrimary** is not called, even though it isESN was not found in the store and it is treating it like a new bot
**_ESN not found in store or this is an associate primary user request, act as if this is a new robot_**

I tried to follow the logic in server.go but my understanding of the code is limited.

Is there something that I'm missing or should all Vectors need to have their user preferences and possibly re-flashed to get the sdK-config.ini to be written correctly.

Thanks

kercre123, 2023-05-20:
If wire-pod had never seen the bot before, it should try to get the bot's SDK certificate from the Anki server, then create an sdk_config.ini entry with that. I'll look into it

bliteknight, 2023-05-21:
> If wire-pod had never seen the bot before, it should try to get the bot's SDK certificate from the Anki server, then create an sdk_config.ini entry with that. I'll look into it

Thank you for looking into this when you can. 

One other thing I noticed: so I factory reset / cleared data / re-activated  two bots so the server created and added both to the sdk_config file. I was able to use the go wrapper for the sdk and communicate with both bots.

When the second bot's IP address changed, the logic somewhere updated the first entry's bot IP address to the new one, so I was no longer able to communicate with either bots because their IP address had changed.

codaris, 2023-10-07:
> If wire-pod had never seen the bot before, it should try to get the bot's SDK certificate from the Anki server, then create an sdk_config.ini entry with that. I'll look into it

It seems like the certificate server is offline now.  Is a certificate necessary for connecting to a Wirepod bot?

closed 2024-05-28 by kercre123