kercre123/wire-pod #143: intent_system_noaudio

radumamy, 2023-05-24:
First, I would like to thank you for this amazing project. Second, I am looking for some advice.

I've installed wire-pod in a Proxmox lxc Debian container and activated a bot today. All seems ok, but I am getting the below message in logs. I've selected VOSK as STT and Houndify as knowledge graph. For some reason, it doesn't understand questions like 'who is barack obama' using the Wikipedia domain.

Can someone please advise how I can fix this? Thank you.

`2023.05.24 11:48:16: Intent matched: intent_play_fistbump, transcribed text: 'fist bump', device: 00702249
2023.05.24 11:48:37: Intent matched: intent_system_noaudio, transcribed text: 'who is barack obama', device: 00702249`

kercre123, 2023-05-24:
With Houndify, you first have to say "hey vector, i have a question", then it will enter into the knowledge graph mode. DDL uses Houndify as both the speech-to-intent and knowledge graph processor, so they were able to add the "intent graph" feature which takes out the "i have a question" step. But, wire-pod only streams audio to Houndify in KG mode, so for now, you still need to say "i have a question" when using KG with wire-pod.

radumamy, 2023-05-24:
Excellent, many thanks.

closed 2023-05-24 by radumamy