kercre123/wire-pod #147: Help me!

khanhnguyen12345, 2023-06-03:
I got an error when VECTOR CONFIGURATION: " error: rpc error: code = Unimplemented desc = unknown service Anki.Vector.external_interface.ExternalInterface ". How to fix it? Thank Admin !

khanhnguyen12345, 2023-06-03:
@kercre123 help me please. Thank You so much!!!

Nanard421, 2023-07-30:
Hello, first thank you ! for creating wire-pod, my vector has re-birth.
 everything seems to work fine except in the Vector Configuration, i have the exact same message.

I'm running server on Windows WSL
Vector (prod version) was previously on firmware 0.9 and is now on 2.0.1 to run wire-pod

DDL servers seems down for months, so could this be the root cause ?
"Error pinging jdocs, likely unauthenticated" : what could that mean ? What is jdocs for ?

[image]

[image]

kercre123, 2023-07-31:
This is a common issue with WSL. I believe it's due to how requests are routed to the Linux VM. It's always best to run it on a native Linux machine, like a Raspberry Pi.

Wire-pod relies on being able to match serial numbers to IP addresses, and using that for identification. However, all requests to wire-pod are routed through the same virtualized network, so they aren't getting the same IP they would need in order to be able to communicate with the robot. So the robot is able to talk to it, but it can't talk to the robot.

Eventually, wire-pod will have a native Windows build which will solve this.

Prolity, 2023-08-14:
I hated this error and fixed it on windows by installing ubuntu on a virtual machine (used VMware). Confirming it is related to wsl. Works 100% as intended

Nuiana, 2023-10-02:
How did you get the VM to run as a bridged network? I can't get mine to do that.. and it can't find it when I do NAT - how did you solve this?

Prolity, 2023-10-02:
Think it depends on vmware version, but bridged connection option was available for me. Maybe it helps to checks the steps on https://thesecmaster.com/install-ubuntu-linux-on-vmware-workstation/

closed 2023-11-26 by kercre123