kercre123/wire-pod #149: Wir-pod Vector connecting issues.

ISCoding1, 2023-06-05:
I have a vector 1.0 production running 2.0.1 but when I try to active it won't work. I have avahi-daemon installed and dbus, avahi-daemon, and wire-pod are all working when I run status. I'm running wire pod on a chromebook acer r11 through the Debian terminal and can access the setup screen through localhost:8080 and through the Ip of wire-pod but not through escapepod.local:8080. Also, I can't activate vector at https://keriganc.com/vector-epod-setup/html/main.html on step 3 when u hit activate after u connect via Bluetooth. What do I do?

ISCoding1, 2023-06-05:
Also the connection usually times out instead of immediately failing to connect, but idk much about coding and how to run this type of stuff.

bliteknight, 2023-07-14:
You have many layers to fix - I do not know if chrome allows you to set the hostname, but that is your first hurdle that you need to overcome. If you cannot access http://escapepod.local:8080 or http://escapepod:8080 on your network, then neither can you Vector, so it wont work. If you figure that out and remove any firewall on your Debian sub os, then you should be godo to go

uandco, 2023-08-16:
@ISCoding1 you need to install Avahi on Debian to enable mDNS and get the .local hostname. See https://wiki.debian.org/Avahi

closed 2023-11-06 by kercre123