kercre123/wire-pod #158: Issue pairing vector on website linked.

plntrm, 2023-06-20:
I might be doing somewthing wrong but everytime I attempt to link vector using bluetooth to the given website, i pair him and the website just goes back to the pair with vector buton?

BedlessBlade, 2023-06-28:
Do you have the bot in recovery mode? If not, recovery mode can be accessed by, when he is on, holding down the power button till it turns off and then keep holding it till a light lights up on his back. You then want to go to that site and, when the bot is on the charger (this is important), press the top of the button on his back twice. This will put him into pairing mode.

Also, if you are using a linux system on that site you actually can't, it must be windows based. This is because a stupid thing with the web bluetooth thing.

Hope this helped!

frien137, 2023-06-29:
Just select paired Vector again. Also check permissions for site in your browser. If using linux check Chrome flags.

closed 2023-11-06 by kercre123