kercre123/wire-pod #160: Is it possible to make the knowledge graph/"Hey Vector, I have a question.." go to a search engine rather than an AI service?

jeffyjeffy1023, 2023-07-04:
I would rather have Vector (or, more specifically, the wire-pod) look towards a search engine and return the top/first result to a query rather than get confidently wrong answers from, say, OpenAI. I'd rather be using Bing of all things. Is that possible, or no? I DO plan on getting a Vector in the future.

Also, some unrelated questions that I'm curious about:
1. Can Vector still take pictures when asked? If so, does he save them to the wire-pod so I can get them from the webpanel?
2. Will bluetooth pairing via the command line ever be added, or no? Since some people's servers might already have bluetooth (as does mine, in addition to Wi-Fi alongside Ethernet for the actual LAN/WAN connection) and it might be more convenient for some to just do it through the terminal. I realize the current BT method is also to push the Escape Pod FW to Vector over web bluetooth, but since that's not available everywhere (i.e. user uses firefox or otherwise can't access web bluetooth) it'd probably be nice to have some kind of python script to do it too maybe..

kercre123, 2023-07-14:
Houndify is a free option which produces accurate answers.

Vector saves pictures to his internal storage, which wire-pod can get through his API and show via the web interface.

I implemented a Go implementation of wire-pod BLE setup, but ended up removing it because of how unstable Go BLE is (it commonly causes the hci device to error out and force a reboot of the system in order for it to work again). It is possible, and works, but the web implementation is much more stable. I could look into making a Python script for it. I would probably need to write a lot of that code on my own though... and its a lot to implement,

jeffyjeffy1023, 2023-07-15:
I see.. I realize that Go is quite the beast to tame, but still, having some sort of easy "send query to external script and wait for feedback" option would be nice, since an external python script could fetch the top/first non-ad result from a google/bing/duckduckgo search, convert the result into text, then feed it back to wire-pod for Vector to read out.

kercre123, 2023-07-19:
That is essentially what Houndify does. It's hard to get just a sentence or two from a search engine which gives all of the information you need to know about a subject. Houndify does that for you.

closed 2023-11-06 by kercre123