kercre123/wire-pod #162: trouble with final step of connecting vector to wirepod

2WR3505, 2023-07-27:
i can link to it by bluetooth using my tablet, i can get it connected to wifi and update to 2.0.1, i just can't get it activate on my wirepod server.

host system is a fresh LMDE 5 install, device used to update him is the Fire HD 10 2019. it's the only thing that seems to work except for the final step.

any suggestions?

uandco, 2023-08-01:
I had the same issue with https://keriganc.com/vector-wirepod-setup/

It didn't download the wirepod firmware after pairing Vector. Maybe there's an issue there where it fails to detect Vector runs a regular firmware which will not be able to connect to `escapepod.local`

To solve the issue, I performed a firmware reset (hold Vector's button till it turns off and the light come up again on the button). You will end up with firmware 0.9 and the welcome screen will mention anki.com.

Just go back to https://keriganc.com/vector-wirepod-setup/, pair again, setup the wifi, then it will install the right firmware and activation will work.

eswstone, 2023-08-28:
I ran into the same issue as 2wr3505, accept i'm using a Pi4 and my motorola cell phone to connect him (they are all on the same wifi network):

First I tried downloading the firmware but it didn't work b/c i didn't know he had to be in recovery mode (blue lights on the back during boot.)

Then I downloaded the firmware successfully onto my vector (blue lights this time) and then set up the vectorx image on the Pi4. But vector could not connect with the Pi (the circle with the head won't turn green)

Then I reset him (memory wipe, back to "ddl.ov" screen) and downloaded the firmware onto my vector AFTER setting up the Pi4 (i made sure the blue lights had come on first). Again it would not connect.

I'm getting the green checkmarks on the vectorx diagnostic screen for the server setup, so I assume it's good. And he's got 2.0.1.6076ep running, i'm pretty sure that's the right firmware.

Any suggestions?

uandco, 2023-08-28:
@eswstone can you access http://escapepod.local/ from a computer on the same network as your Vector? That's the host an `ep` firmware would try to connect to.

eswstone, 2023-08-29:
@uandco soooooo i figured out that the setup is, for some reason, it's starting chipper to port 8040 for some reason. I assume it's because 8080 is being used by another ip address but i'm still learning how all that works. I was able to determine that, for a moment there, port 8080 was associated with 10.0.0.190 so i just changed the IP addy in the Pi but that only worked for a few seconds before i wouldn't connect with the escapepod.local:8080
all that tells me i'm doing it the right way, there's just some networking tomfoolery I have to figure out. My other problem is my internet provider has port forwarding locked out from me so i can't change anything. 
OH, I also tried using my phone's hotspot to try and onboard him but once i got his firmware updated, he powered back on but would not fully initialize (head didn't lift, screen face stayed blank). Do you know how to change his wifi info after he's downloaded his firmware but BEFORE he's been onboarded?

uandco, 2023-08-29:
If 8080 is used, it’s used by the Pi itself, not by another host on the network so changing the IP of the wire-pod host will not achieve anything.

Also not getting the question about port forwarding. It all happens local. As long as both Vector and the wire-pod server are on the same local network, it’s going to find escapepod.local.

eswstone, 2023-08-31:
@uandco that makes sense, thanks for clearing that up. 

I figured it out! I wasn't wearing my glasses while sticking just the tip of my tongue out in a westerly direction! No, i really have no clue what made it work but it did. I'm running off the vectorx flash, installed the firmware on him using my cellphone, can't say I did anything different this time. Hope I don't have to go through that again! 

Thank you for your help, wisdom and support. Now i'm off to troubleshoot vectorx commands while learning to write my own. LONG LIVE VECTOR!!!!

KenwoodFox, 2023-10-03:
I'm so confused, escapepod.local resolves fine, but my bot seems totally lost, unable to find it. just constant `The bot is likely unable to communicate with your wire-pod instance.`

uandco, 2023-10-03:
> I'm so confused, escapepod.local resolves fine, but my bot seems totally lost, unable to find it. just constant `The bot is likely unable to communicate with your wire-pod instance.`

Make sure the bot has an `ep` firmware installed, that's the only firmware that will try to connect to `escapepod.local` instead of the cloud server. See my comment above: https://github.com/kercre123/wire-pod/issues/162#issuecomment-1660020627

KenwoodFox, 2023-10-03:
> > I'm so confused, escapepod.local resolves fine, but my bot seems totally lost, unable to find it. just constant `The bot is likely unable to communicate with your wire-pod instance.`
> 
> Make sure the bot has an `ep` firmware installed, that's the only firmware that will try to connect to `escapepod.local` instead of the cloud server. See my comment above: [#162 (comment)](https://github.com/kercre123/wire-pod/issues/162#issuecomment-1660020627)

I think I've done that reset procedure a couple times before, how can i check what firmware I'm using? and, if i just reset/erase it, how do i make sure the tool actually loads the new firmware? is there an option in the setup for that after pairing?

uandco, 2023-10-03:
> > > I'm so confused, escapepod.local resolves fine, but my bot seems totally lost, unable to find it. just constant `The bot is likely unable to communicate with your wire-pod instance.`
> > 
> > 
> > Make sure the bot has an `ep` firmware installed, that's the only firmware that will try to connect to `escapepod.local` instead of the cloud server. See my comment above: [#162 (comment)](https://github.com/kercre123/wire-pod/issues/162#issuecomment-1660020627)
> 
> I think I've done that reset procedure a couple times before, how can i check what firmware I'm using? and, if i just reset/erase it, how do i make sure the tool actually loads the new firmware? is there an option in the setup for that after pairing?
> 
> Ah, i see, ok, it seems i don't have the `ep` firmware [image] But no amount of resetting/erasing seems to get it to load? is there a manual way?

https://keriganc.com/vector-wirepod-setup/ will load the `ep` firmware after pairing if Vector has just been factory reset.

KenwoodFox, 2023-10-03:
> > > > I'm so confused, escapepod.local resolves fine, but my bot seems totally lost, unable to find it. just constant `The bot is likely unable to communicate with your wire-pod instance.`
> > > 
> > > 
> > > Make sure the bot has an `ep` firmware installed, that's the only firmware that will try to connect to `escapepod.local` instead of the cloud server. See my comment above: [#162 (comment)](https://github.com/kercre123/wire-pod/issues/162#issuecomment-1660020627)
> > 
> > 
> > I think I've done that reset procedure a couple times before, how can i check what firmware I'm using? and, if i just reset/erase it, how do i make sure the tool actually loads the new firmware? is there an option in the setup for that after pairing?
> > Ah, i see, ok, it seems i don't have the `ep` firmware [image] But no amount of resetting/erasing seems to get it to load? is there a manual way?
> 
> [keriganc.com/vector-wirepod-setup](https://keriganc.com/vector-wirepod-setup/) will load the `ep` firmware after pairing if Vector has just been factory reset.

I think I've got it, there's a difference between factory reset and just, regular menu reset! It did do the firmware update but, i still don't think its connected, its still flashing the app symbol and everything, im tracking on #198 so i don't flood 2WR3505 if that's ok?

closed 2023-11-06 by kercre123