kercre123/wire-pod #163: Vector changed IP and cannot be reached anymore

MTrab, 2023-07-30:
My 2 Vectors got a new IP address and not Wire-Pod cannot connect to the Vectors to do configurations.
Where can I change the IP on the robots in Wire-Pod?

MTrab, 2023-07-30:
Found and changed in `~/wire-pod/chipper/jdocs/botSdkInfo.json` and Wire-Pod works after a restart.
Could this perhaps be changed so that Wire-Pod identifies by serial and not IP?

closed 2023-07-30 by MTrab

kercre123, 2023-07-30:
Due to a workaround I had to implement (involves working around per-robot keys), there is a point where wire-pod has to rely on a request from the robot which doesn't include the bot's serial number. It just knows that it is from a robot, and it knows the IP address of the request. I very much recommend setting a static IP for everything. I can include that in the documentation

MTrab, 2023-07-30:
I know it's recommend, just did some LAN maintenance including moving some DHCP reservations when this happened ;)

But yeah, might be worth noting in the documentation as a recommendation