kercre123/wire-pod #164: Suggestions to make the setup work on MacOS

uandco, 2023-08-01:
Hi,

I managed to run wire-pod on MacOS with a few tweaks. It would be great to have them incorporated to the repo.

in `setup.sh`, I had to replace `/root` with `$HOME` in multiple places as there's no `/root` folder in MacOS, and you can't easily create a new folder in `/` in recent OSes.

As mentioned in the script itself, you need to install `brew`, `go` and the following `brew` packages:
```
brew install libopusenc
brew install libsodium
brew install opusfile
```

The latest `vosk-api`release doesn't contain precompiled darwin assets so you have to manually install https://github.com/alphacep/vosk-api/releases/download/v0.3.42/vosk-osx-0.3.42.zip which is the most recent osx asset.

Once downloaded, copy `vosk_api.h` to `~/.vosk/libvosk` (if you ran the setup from your user directory).
Copy `libvosk.dylib` to `/usr/local/lib`
Right click and choose "open" on `libvosk.dylib` to allow MacOS to allow the file to be open or it will fail when accessed later.

Failing to replace the header file and use the osx compiled lib will produce this error when running `sudo ./chipper/start.sh`:

```
ld: warning: ignoring file /Users/thibaut/.vosk/libvosk/libvosk.so, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 )
```

For some reason I also had to copy some files around to correct this issue:

```
ld: warning: directory not found for option '-L/Users/thibaut/go/pkg/mod/github.com/alphacep/vosk-api/go@v0.3.45/../src'
```

I copied the content of `go/pkg/mod/github.com/alphacep/vosk-api@v0.3.45` into `go/pkg/mod/github.com/alphacep/vosk-api` so it could find the `src` folder.

Last but not least, even when setting your computer name to `escapepod`, it will still complain it's incorrect as MacOS reports the name as `escapepod.local`:

```
WARNING: You have chosen the Escape Pod config, but the system hostname is not 'escapepod'. This means your robot will not be able to communicate with wire-pod unless you have a custom network configuration.
Actual reported hostname: escapepod.local
```

Ignore this, `escapepod.local` works fine.

I hope this helps others trying to run this on MacOS and some changes can be made to the scripts / wiki.

lcxmas, 2023-08-14:
I am desperate.
so, please be patient.....and help me.
i did the following:

Installed :
https://github.com/alphacep/vosk-api/releases/download/v0.3.42/vosk-osx-0.3.42.zip w

copy vosk_api.h to ~/.vosk/libvosk (if you ran the setup from your user directory).
Copy libvosk.dylib to /usr/local/lib
Right click and choose "open" on libvosk.dylib to allow MacOS to allow the file to be open or it will fail when accessed later.

**I can't find   ~/.vosk/libvosk** 

i am stuck here

lcxmas, 2023-08-14:
i have no idea where i could find this:
go/pkg/mod/github.com/alphacep/vosk-api@v0.3.45

uandco, 2023-08-14:
@lcxmas the go folder would also be in your user directory if you started the setup from there, so ~/go/pkg/mod/github.com/alphacep/vosk-api@v0.3.45

lcxmas, 2023-08-15:
Thanks for your explanation.
compiling is not my bread and butter but i am not totally a stranger to terminal and basic compiling.
having said that.
i really don't have:
~/go/pkg/mod/github.com/alphacep/vosk-api@v0.3.45
[image]

About this, ~/.vosk/libvosk .
in my .vosk there was no libvosk folder. i had to create one

uandco, 2023-08-15:
That’s weird, looks like it’s not downloading the same packages as on my machine. 

A full output of both setup.sh and start.sh would help I guess.

lcxmas, 2023-08-15:
Definitely struggling, now i am getting errors from nowhere.
i really need an official support!

uandco, 2023-08-16:
@lcxmas try `brew install pkg-config` before running `start.sh`

lcxmas, 2023-08-16:
doing again now!

lcxmas, 2023-08-21:
i have got this error:

escapepod:wire-pod luca01$ sudo ./chipper/start.sh
# command-line-arguments
/usr/local/go/pkg/tool/darwin_amd64/link: running clang failed: exit status 1
ld: warning: ignoring file /Users/luca01/.vosk/libvosk/libvosk.so, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 )
Undefined symbols for architecture x86_64:
  "_vosk_gpu_init", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_gpu_init in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_gpu_init)
  "_vosk_gpu_thread_init", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_gpu_thread_init in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_gpu_thread_init)
  "_vosk_model_find_word", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_model_find_word in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_model_find_word, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_model_find_word )
  "_vosk_model_free", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_model_free in 000003.o
     (maybe you meant: _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_model_free, __cgo_b69c6a807b28_Cfunc_vosk_model_free )
  "_vosk_model_new", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_model_new in 000003.o
     (maybe you meant: _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_model_new, __cgo_b69c6a807b28_Cfunc_vosk_model_new )
  "_vosk_recognizer_accept_waveform", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_accept_waveform in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_accept_waveform, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_accept_waveform )
  "_vosk_recognizer_final_result", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_final_result in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_final_result, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_final_result )
  "_vosk_recognizer_free", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_free in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_free, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_free )
  "_vosk_recognizer_new", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new_grm, __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new , _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_new , __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new_spk )
  "_vosk_recognizer_new_grm", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new_grm in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new_grm)
  "_vosk_recognizer_new_spk", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new_spk in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_new_spk)
  "_vosk_recognizer_partial_result", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_partial_result in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_partial_result, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_partial_result )
  "_vosk_recognizer_reset", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_reset in 000003.o
     (maybe you meant: _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_reset, __cgo_b69c6a807b28_Cfunc_vosk_recognizer_reset )
  "_vosk_recognizer_result", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_result in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_result, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_result )
  "_vosk_recognizer_set_max_alternatives", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_max_alternatives in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_max_alternatives, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_max_alternatives )
  "_vosk_recognizer_set_partial_words", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_partial_words in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_partial_words, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_partial_words )
  "_vosk_recognizer_set_spk_model", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_spk_model in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_spk_model, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_spk_model )
  "_vosk_recognizer_set_words", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_words in 000003.o
     (maybe you meant: _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_words, __cgo_b69c6a807b28_Cfunc_vosk_recognizer_set_words )
  "_vosk_set_log_level", referenced from:
      __cgo_b69c6a807b28_Cfunc_vosk_set_log_level in 000003.o
     (maybe you meant: __cgo_b69c6a807b28_Cfunc_vosk_set_log_level, _github.com/alphacep/vosk-api/go._cgo_b69c6a807b28_Cfunc_vosk_set_log_level )
  "_vosk_spk_model_free", referenced from:
[... cut ...]
6. **Library Versions:** It could be that the version of `libvosk.so` is not compatible with your program's version. Ensure that library versions are consistent and compatible.

To resolve the issue, you might want to carefully check the linking options, ensure that the libraries have been compiled correctly for your architecture and OS version, and verify that the library path is correct. If the problem persists, you could seek further resources on the Vosk library support forum or consult development experts for additional assistance.

What do you think?

uandco, 2023-08-21:
It means you didn’t replace libvosk.h and it’s trying to link the libvvosk.so lib instead of libvosk.dylib

lcxmas, 2023-08-21:
replace libvosk.h with what and ...where?

uandco, 2023-08-21:
From my original post:

Once downloaded, copy vosk_api.h to ~/.vosk/libvosk (if you ran the setup from your user directory).

lcxmas, 2023-08-21:
i think i was able to fix this issue.

what about this error:

ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

uandco, 2023-08-21:
Not sure about that one, are you compiling on a Mac Silicon or Intel?

lcxmas, 2023-08-22:
> Not sure about that one, are you compiling on a Mac Silicon or Intel?

Mac intel

uandco, 2023-09-01:
I was also compiling on Mac Intel, this error should go away once you use the correct vosk header file and lib.

closed 2023-11-06 by kercre123