kercre123/wire-pod #168: Way to change required hostname?

PascalHesselink, 2023-08-14:
So, I have everything installed now.

The only problem is that the connections requires the "escapepod.local" and since I have a Fritzbox I'm unable to change the extension/domain. So I only can do "escapepod.fritz.box"

Is there ANY way to tell the configuration to look for a different domain? 
*Without installing PiHole.*

kercre123, 2023-08-14:
`escapepod.local` is hardcoded in the bot firmware created by DDL, and it cannot be changed. The only way around this is with a router which uses `.local`, or by setting up some way to route it to the IP. Any reason you are against using PiHole?

uandco, 2023-08-16:
If your device uses mDNS (multicast DNS) it should also have the .local hostname. What platform are you running it off?

closed 2023-11-06 by kercre123

hhauschild, 2023-12-22:
I had the same problem with a Fritzbox 6591 and escapepod on a Raspi, and properly using mDNS was the only way to solve it. The domain ".local" seems to be reserved for mDNS and Vector seems to honor the specification. So if your hostname is already "escapepod", then just run the two "avahi-deamon" commands mentioned in https://github.com/kercre123/wire-pod/wiki/Installation and maybe restart wire-pod and Vector a few times. 
Using pi-hole is not going to help you here. You can configure a local domain "escapepod.local" and it will be resolved on a Windows computer and some devices, but not by Vector. At least not by my Vector...
Also config changes in the Fritzbox wont get you any further, for the same reason: they all update DNS resolution but ".local" is not meant to be resolved by normal DNS.

PascalHesselink, 2023-12-22:
> I had the same problem with a Fritzbox 6591 and escapepod on a Raspi, and properly using mDNS was the only way to solve it. The domain ".local" seems to be reserved for mDNS and Vector seems to [...]

Hi, since the beginning of August I have been using pi-hole to connect Vector, it still works fine. So it definitely works since it was unable to connect before using pi-hole. It's also possible to create a hotspot from the raspberry pi and connect the vector to that. Creating a proxy for the local domain escapepod.local resulted in a stable connection.