kercre123/wire-pod #169: Self hosting vector-wirepod-setup and questions

Neo-dessinator, 2023-08-16:
hello there !
I am discovering this and I am a bit impatient to tests things =), but I have several questions.

- is there a way to self host the  vector-wirepod-setup web app ? just to be able to run all things locally (I am a programmer I can tweak things if necessary). It seems to serve the same purpose as the escapepod launch. but with more features
- does the vector-wirepod-setup install the same firmware as the escapepod ? if that so, during the installing process, is it possible to avoid this step ?
- is it possible to run both escapepod, wire-pod, and vectorX on the same RPI4 ? (maybe by tweaking the deployment address and port)

I have an escape pod, which work but I feel scammed because of the lack of documentation, and the escapepod-extension is a total pain to make it work (never succeed yet ...)

kercre123, 2023-08-20:
vector-wirepod-setup is a static webpage, taken from DDL's vector-setup.ddl.io page (down now). You can use HTTrack or similar software to download it.

The firmware is the same as the escape pod firmware. So yes, you can skip the step. Though, you will still need the page for authorizing the robot with your wire-pod instance.

You should be able to run all of them if you tweak things.

Neo-dessinator, 2023-08-29:
Thanks for your kind reply.
Sorry to have rised an issue instead of a discussion. I wasn't aware of them :/

I am starting to see a bit more how it does work.
I never thought that it will be so easy to pass thorugh the authenticate process. 
So if I understood well, the process is that once authenticated, the RTS login looks for a something (a bot) , is that so ?

I notice that the session json returned by the session endpoint is always the same, ( at least at first sight). The expiration date is In november 2023. It could be extended by just changing the date to another farther in the futur =).

I hope to be able to help on this project .

Thanks for your time !

closed 2023-11-06 by kercre123