kercre123/wire-pod #180: Vector Microphones @ -1000

franko-c, 2023-09-02:
Hi all,

I am super grateful for this project and I'm doing what I can to make sense of everything.

My Vector is successfully connected to wire-pod and I can configure them from the web interface! 

Unfortunately, I still have some issues:

- Vector will not speak 
- Vector will not do anything of their own free will (I am currently issuing commands from the web interface)
- Acknowledges his cube when commanded to, but never plays with it
- Microphones (as viewed in the CCIS interface) read a constant -1000, only budging when blown into or shouting, as a result, limited to no voice commands. I cannot get my name set up
- pc error: code = DeadlineExceeded desc = context deadline exceeded after attempting voice commands 

Has anyone experienced some of these issues and resolved them? I have already tried a different STT option in the setup, Vector is running v2.0.1.6076 and I never saw an option to modify which firmware it runs but perhaps I missed something, or a lot of things.

Thank you!

Hugo123456767564577547, 2023-09-29:
Put him on his charger, then hold his button down for fifteen seconds until the second light goes deep blue then, we he reboots, go to keriganc.com/vector-wirepod-setup and update his firmware, this won’t clear user data btw

kercre123, 2023-11-06:
This is a hardware issue, not wire-pod related.

closed 2023-11-06 by kercre123