kercre123/wire-pod #181: Vector Configuration Connection Issue

transformers1769, 2023-09-05:
Hi! Thanks for the tutorial. I got Vector to work real easy.

The only thing is, I can't seem to connect to Vector to change some of the settings. The Bot serial number shows up in the drop down but when I click connect, nothing happens. It is connecting and I can activate Vector and change the local time/temp/measure system but that's it

Have you encountered this before and is there a solution?

Thanks
[image]

glowinglightsaber3, 2023-11-12:
Also having this issue, getting 

error: rpc error: code = Unimplemented desc = unknown service Anki.Vector.external_interface.ExternalInterface

edd134, 2023-11-20:
This is to do with wsl, kercre123 just released a windows version of wire pod with probably fixes it (but its broken)